openshwprojects/OpenBK7231T_App #922: Second relay not being reported into HomeAssistant

NielsPiersma, 2023-09-13:
**Describe the bug**
After doing a teardown of the Ledvance Smart Wifi Orbis KITE and successfully flashing it with OpenBeken if found that all works fine from the webgui. However, when using MQTT autodiscovery only LED/RELAY 0 is published and taken into HomeAssistent. I did have some hit-and-miss successes when configuring one switch as a relay and one switch as an LED.  But when I deleted the device in HomeAssistant and reran auto-discovery, only relay / LED 0 popped up.

**Firmware:**
- Version 1.17.244
- Device: LEDVANCE SMART WIFI ORBIS KITE RD 410 TW WT
- Chip/model: BK7231T
- Device config
- P7 (PWM1) LED 1
- P8 (PWM2) LED 0

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Config/Home Assistant  Configuration'
2. Click on 'Start Home Assistant Discovery'
3. Go to Home Assistant devices filer on MQTT 
4. Select OpenBeken device
5. Only one relay led appears
[image]
[image]

**Screenshots**
Screen shots show the issue

Weirdly enough after like 5 or 6 minutes the second relay does appear. Question is, why doesn't the second relay report instantly, like the first one.

Thanks in advance.
Niels

openshwprojects, 2023-09-14:
Hello, what is the device config in OBK JSON?

What does the HA receive when you do the discovery?

NielsPiersma, 2023-09-14:
Not too complicated :) 

{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BK7231T",
  "board": "TODO",
  "flags": "134218752",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "7": "LED;1",
    "8": "LED;0"
  },
  "command": "scheduleHADiscovery 10",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}

And two,
When I delete the device in HA, and execute a discovery, this is what is logged at HA:

[image]

As you can see, there is only one switch.

Rebooting and restarting the HA discovery doesn't place back the missing LED/RELAY 

Any hints appreciated :)

NielsPiersma, 2023-09-14:
Well, workaround found.
Remove the device from HA
and reboot. Now it is added properly...

2023-09-14 20:22:30.501 DEBUG (MainThread) [homeassistant.components.mqtt.client] Transmitting retained message on homeassistant/light/OpenBK7231T_177E1CEA_light_0/config: '', mid: 63460, qos: 0
2023-09-14 20:22:30.502 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homeassistant/light/OpenBK7231T_177E1CEA_light_0/config (qos=0): b''
2023-09-14 20:22:30.528 DEBUG (MainThread) [homeassistant.components.mqtt.client] Transmitting retained message on homeassistant/light/OpenBK7231T_177E1CEA_light_1/config: '', mid: 63461, qos: 0
2023-09-14 20:22:30.528 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homeassistant/light/OpenBK7231T_177E1CEA_light_1/config (qos=0): b''
2023-09-14 20:22:30.660 DEBUG (MainThread) [homeassistant.components.mqtt.client] Unsubscribing from obk177E1CEA/connected, mid: 63462
2023-09-14 20:22:30.660 DEBUG (MainThread) [homeassistant.components.mqtt.client] Unsubscribing from obk177E1CEA/0/get, mid: 63462
2023-09-14 20:22:30.660 DEBUG (MainThread) [homeassistant.components.mqtt.client] Unsubscribing from obk177E1CEA/1/get, mid: 63462
2023-09-14 20:23:02.860 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homeassistant/light/OpenBK7231T_177E1CEA_light_0/config (qos=0): b'{"dev":{"ids":["OpenBK7231T_177E1CEA"],"name":"obk177E1CEA","sw":"1.17.244","mf":"Beken Corporation","mdl":"BK7231T","cu":"http://10.4.3.128/index"},"name":"0","~":"obk177E1CEA","avty_t":"~/connected","pl_on":"1","pl_off":"0","uniq_id":"OpenBK7231T_177E1CEA_light_0","qos":1,"stat_t":"~/0/get","cmd_t":"~/0/set"}'
2023-09-14 20:23:02.860 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'device': {'identifiers': ['OpenBK7231T_177E1CEA'], 'name': 'obk177E1CEA', 'sw_version': '1.17.244', 'manufacturer': 'Beken Corporation', 'model': 'BK7231T', 'configuration_url': 'http://10.4.3.128/index'}, 'name': '0', 'availability_topic': 'obk177E1CEA/connected', 'payload_on': '1', 'payload_off': '0', 'unique_id': 'OpenBK7231T_177E1CEA_light_0', 'qos': 1, 'state_topic': 'obk177E1CEA/0/get', 'command_topic': 'obk177E1CEA/0/set', 'platform': 'mqtt'}
2023-09-14 20:23:02.860 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: light OpenBK7231T_177E1CEA_light_0
2023-09-14 20:23:02.864 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Pending discovery for ('light', 'OpenBK7231T_177E1CEA_light_0'): deque([])
2023-09-14 20:23:02.902 DEBUG (MainThread) [homeassistant.components.mqtt.client] Received message on homeassistant/light/OpenBK7231T_177E1CEA_light_1/config (qos=0): b'{"dev":{"ids":["OpenBK7231T_177E1CEA"],"name":"obk177E1CEA","sw":"1.17.244","mf":"Beken Corporation","mdl":"BK7231T","cu":"http://10.4.3.128/index"},"name":"1","~":"obk177E1CEA","avty_t":"~/connected","pl_on":"1","pl_off":"0","uniq_id":"OpenBK7231T_177E1CEA_light_1","qos":1,"stat_t":"~/1/get","cmd_t":"~/1/set"}'
2023-09-14 20:23:02.903 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Process discovery payload {'device': {'identifiers': ['OpenBK7231T_177E1CEA'], 'name': 'obk177E1CEA', 'sw_version': '1.17.244', 'manufacturer': 'Beken Corporation', 'model': 'BK7231T', 'configuration_url': 'http://10.4.3.128/index'}, 'name': '1', 'availability_topic': 'obk177E1CEA/connected', 'payload_on': '1', 'payload_off': '0', 'unique_id': 'OpenBK7231T_177E1CEA_light_1', 'qos': 1, 'state_topic': 'obk177E1CEA/1/get', 'command_topic': 'obk177E1CEA/1/set', 'platform': 'mqtt'}
2023-09-14 20:23:02.903 INFO (MainThread) [homeassistant.components.mqtt.discovery] Found new component: light OpenBK7231T_177E1CEA_light_1
2023-09-14 20:23:02.906 DEBUG (MainThread) [homeassistant.components.mqtt.discovery] Pending discovery for ('light', 'OpenBK7231T_177E1CEA_light_1'): deque([])
2023-09-14 20:23:03.008 DEBUG (MainThread) [homeassistant.components.mqtt.client] Subscribing to obk177E1CEA/connected, mid: 63468, qos: 1
2023-09-14 20:23:03.009 DEBUG (MainThread) [homeassistant.components.mqtt.client] Subscribing to obk177E1CEA/0/get, mid: 63468, qos: 1

NielsPiersma, 2023-09-14:
Second light has the same issue, also exists with lower firmware:
Build on Aug 29 2023 06:14:35 version 1.17.230