openshwprojects/OpenBK7231T_App #923: OpenBeken rebooting when simultaneously switching off two relays at the same time.

NielsPiersma, 2023-09-13:
**Describe the bug**
When using the Show Header Toggle in HomeAssistant to power OFF two relays at the SAME time, the device will reboot. 
when toggling independently, all works fine.

**Firmware:**
Version 1.17.244
Device: LEDVANCE SMART WIFI ORBIS KITE RD 410 TW WT
Chip/model: BK7231T
Device config
P7 (PWM1) LED 1
P8 (PWM2) LED 0
FLAGS: 134218752
 
**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'HomeAssistant'
2. Toggle the Header above the device with two relays
3. See ping disapear

**Screenshots**

[image]
[image]

**Additional context**
Maybe this can be solved by adding some delay for switching the secondary relay.
There is nothing logged in the log it silently reboots.
Only found it because I needed to power the lights on again after turning them of by accident.

openshwprojects, 2023-09-14:
Hello, do you have PowerSave 1 (enabled low power mode)?

NielsPiersma, 2023-09-14:
> Hello, do you have PowerSave 1 (enabled low power mode)?

No, I've not enabled PowerSave 1, tried explicitly with "PowerSave 0", but to no avail.

It instantly reboots when toggling both switches simultaneously.

But I think it must have something to do with the power drawn, as when I configure a bogus (chip only on a breadboard) with two relays or two leds I can switch them off both simultaneously without any problem.

The workaround is not that difficult, adding a few ms of delay in the automation and not using the header toggle keeps the device from rebooting.

Niels

g6094199, 2024-04-17:
this sound like a power problem to me. while activating 2 relays the voltage drops to much an the device reboots.