openshwprojects/OpenBK7231T_App #924: IR driver on BL602

rnrobles, 2023-09-15:
**Describe the bug**
When trying to use the IR module as a receiver, apparently it does not have that driver enabled with the BL602 chip, is it possible to add it?

**Firmware:**
- Version 1.17.244
- Device ZJ-bwcd-ir-vc MagicHome
- Chip/model: BL602
- Device config: 
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BL602",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "3": "PWM;1",
    "4": "PWM;0",
    "12": "IRRecv;1",
    "21": "PWM;2"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}

[image]
[image]

openshwprojects, 2023-09-16:
IR interrupt is not implemented on BL602, that would require some programming work, do you know C?

rnrobles, 2023-09-16:
a few years ago I used a bit with arduino, if you guide me maybe I could help with that part

normanr, 2023-11-27:
Note that the BL602 has a built-in IR remote HW accelerator, but it's buffer is only large enough to receive commands from "simple" remotes, eg for TV's, etc, that only send 4 bytes of data. More complex remotes, eg those used by AC systems, can send 12 bytes, which will not fit into the HW accelerator buffer.

I have had good success reading the longer signals by using the IR receiver as a regular GPIO pin that triggers an interrupt and then just tight loop reading the pin until all the data has been received.