openshwprojects/OpenBK7231T_App #925: Tasmota FANSPEED command not found

n8man, 2023-09-18:
**Describe the bug**
Tasmota supports a `FANSPEED` command which doesn't appear to exist in OpenBeken... But maybe I'm doing this wrong.

**Firmware:**
- Version [e.g. 1.17.245]
- Device: Prichor RC05+5R Smart WiFi Ceiling Fan Remote Control (https://www.amazon.com/dp/B0C4X7S3RR)
- Chip/model: BK7231N CB2S
- Device config:
```
startDriver TuyaMCU
tuyaMcu_setBaudRate 9600

// fan switch dpId 60 bool (type 1)
setChannelLabel 0 FanSwitch
setChannelType 0 toggle
linkTuyaMCUOutputToChannel 60 1 0

// fan speed dpId 62 enum (type 4)
setChannelLabel 1 FanSpeed
setChannelType 1 OffLowMidHigh
linkTuyaMCUOutputToChannel 62 4 1

// light switch dpId 20 bool (type 1)
setChannelLabel 2 Light
setChannelType 2 toggle
linkTuyaMCUOutputToChannel 20 1 2

// light dimmer dpId 22 value (type 2)
setChannelLabel 3 Dimmer
setChannelType 3 dimmer
linkTuyaMCUOutputToChannel 22 2 3
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to /cmd_tool
2. Run command `FANSPEED 1`
3. See error `CMD:cmd FANSPEED NOT found (args 2)`

openshwprojects, 2023-09-18:
Hello, I think that with our new flexible dpID mapping system, you can just use:
setChannel 3 2
etc etc, to set channel 3 which you have mapped to fan speed...

This is a very flexible approach because channels are like variables and are fully scriptable

n8man, 2023-09-20:
@openshwprojects that's true. The issue was that I was trying to use a tasmota "driver" within Hubitat which appeared to rely on the FANSPEED command. I can hack up the driver to use setChannel, but then hub/driver would need to know the implementation detail of what channel is being used to specify the fan speed. I was hoping that openbeken supported FANSPEED and that I just needed to do some kind of mapping on the device to keep that implementation detail (channel) hidden from the hub.

Not a big deal at all though.

Is the DIMMER command implemented?

openshwprojects, 2023-09-20:
Ah, I see, you need it for Hubitat... No problem there, I can implement those commands if they are needed by a third party device. 

Let me know what you need and what kind of arguments should it take and I can add it to the codebase for you.