openshwprojects/OpenBK7231T_App #928: send MQTT request via SendGet command in autoexec.bat

Wolfieeewolf, 2023-09-25:
Is it possible to send MQTT request via the SendGet command?

I use Zigbee2MQTT and it would be great if I could send an MQTT request to turn off a light when I press the button on my OBK switch.

something like

`addEventHandler OnClick 8 SendGet cmnd/OusideLight/PowerToggle`

openshwprojects, 2023-09-25:
Well, aren't you looking for a Publish command?

Wolfieeewolf, 2023-09-25:
I guess so.

I'm not really familiar with MQTT commands.

Can OBK already do something like this?

openshwprojects, 2023-09-30:
You can do a publish in obk from script and in HA watch for it to trigger any automation you want

Wolfieeewolf, 2023-10-01:
I tried something like this

`addEventHandler OnClick 10 publish zigbee2mqtt/lounge/set/TOGGLE`

I can see it registering in MQTT Explorer but it doesn't turn the light on or off.

I'm not very good with MQTT commands and could very well have it wrong.  I have done some searching but haven't found anything that works yet.

openshwprojects, 2023-10-02:
What is the expected Zigbee2MQTT publish syntax? What is the topic and payload you want to send?

Wolfieeewolf, 2023-10-03:
as per this Zigbee2MQTT site the publish syntax should be  zigbee2mqtt/FRIENDLY_NAME/set with payload {"state": "TOGGLE"}
[Zigbee2MQTT Ikea light](https://www.zigbee2mqtt.io/devices/LED1622G12.html)

Following the [MQTT topics page](https://www.zigbee2mqtt.io/guide/usage/mqtt_topics_and_messages.html) it should be correct but doesn't trigger the light.

Could be because it's using JSON commands. I'm not sure if OBK supports the sending of this data.

I can see that it triggers the light in MQTT Explorer but the command is not correct so doesn't turn on the light. The syntax is correct as far as I can tell.

Wolfieeewolf, 2023-10-20:
> That is possible, but you must use a third parameter with the value of "-1" on the publish command. For example: publish zigbee2mqtt/FRIENDLY_NAME/set Toggle -1

This works perfectly :)

closed 2023-10-20 by Wolfieeewolf