openshwprojects/OpenBK7231T_App #929: Build of OpenBK7231N fails if ENABLE_LITTLEFS is undefined

arnonm, 2023-09-27:
**Describe the bug**
undefining ENABLE_LITTLEFS in obk_config.h (as an effort to safe space) causes src/httpserver/reset_interface.c to fail compilation as LFS_BLOCKS_END is not defined

**Firmware:**
- Version latest branch
- Device - NA
- Chip/model: BK7231N
- Device config  - NA

**To Reproduce**
Steps to reproduce the behavior:
add:
1. #undef ENABLE_LITTLEFS in line 58 of obk_config.h
2. build source
Build fails

openshwprojects, 2023-09-30:
Hello, I have fixed it in the last commit, can you check? Use #define ENABLE_LITTLEFS 0

arnonm, 2023-09-30:
httpserver/reset_interface.c still failed on LFS_BLOCKS_END
This workaround resolved it, but I don't know if it is valid 

if PLATFORM_BK7231T
		return http_rest_post_flash(request, START_ADR_OF_BK_PARTITION_OTA, LFS_BLOCKS_END);
#elif PLATFORM_BK7231N
#if ENABLE_LITTLEFS
		return http_rest_post_flash(request, START_ADR_OF_BK_PARTITION_OTA, LFS_BLOCKS_END);
#else
		return http_rest_post_flash(request, -1, -1);
#endif

openshwprojects, 2023-10-01:
are you sure? it works for me without workaround:

https://github.com/openshwprojects/OpenBK7231T_App/pull/932