openshwprojects/OpenBK7231T_App #934: IP obtaining timeout

Thor368, 2023-10-03:
**Describe the bug**
Device connects to WiFi but times out after a few seconds and claims: IP obtaining timeout. Even tho the DHCP server of that network is up and running fine. Setting a static IP results in the same behaviour. Log file follows below.

**Firmware:**
- Versions tested: 1.17.262 and 1.17.0
- Device: Magic Home RGBW controler
- Chip/model: FL868C20 (BL602)

**To Reproduce**
1. Flash device
2. Connect to captive portal and set WiFi credentials
3. Device loops while trying to optain an IP

**LOG**
[WF][PF] Using profile, idx is @0
[WF][SM] Retry Again --->>> retry connect
[WF][SM] Exiting disconnect state
Delete Timer.
[WF][SM] State Action ###disconnect### --->>> ###connecting###
[WF][SM]  Action Connect
           ssid ExampleAP
           ssid len 7
           psk abcd1234
           psk len 8
           pmk 
           pmk len 0
           mac 00:00:00:00:00:00
connecting using vif_idx 0
===start sm_get_bss_params===
bssid[0] = 0xffff
bssid[1] = 0xffff
bssid[2] = 0xffff
search ssid = ExampleAP
result ssid index = 0
===end sm_get_bss_params===
[WF][SM] Entering connecting state
[APP] [EVT] Reconnect 447732
[APP] [EVT] Reconnect 447732
[APP] [EVT] Connecting 447732
[APP] [EVT] Connecting 447732
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_CONNECTING - 1
wpa2/wpa3 pairwise ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 group ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 is_pmf_required:0
AP Security mode: wep:0,wpa:0,wpa2:1,wpa3:0
Probe Response
timestamp = 0x4c8c04d4b
wpa2/wpa3 pairwise ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 group ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 is_pmf_required:0
AP Security mode: wep:0,wpa:0,wpa2:1,wpa3:0
wpa2/wpa3 pairwise ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 group ccmp:1,tkip:0,wep104:0,wep:40:0
wpa2/wpa3 is_pmf_required:0
AP Security mode: wep:0,wpa:0,wpa2:1,wpa3:0
------ 247 set default key 0x4201dfb0, key ptr 0x00000000
[17:20:44.346] - [APP] [EVT] SCAN On Join 448242
[APP] [EVT] Unknown code 10, 448242
set key:sta_idx:1, inst_nbr:0, key_idx:0,cipher_suite:2,key_len:16
set key:sta_idx:255, inst_nbr:0, key_idx:1,cipher_suite:2,key_len:16
ind ix 0x4202a584, chan_ctxt is 0x4201f22c
[RX] Connection Status
[RX]   status_code 0
[RX]   connect result: sm connect ind ok
[RX]   MAC xx:xx:xx:xx:xx:xx
[RX]   vif_idx 0
[RX]   ap_idx 1
[RX]   ch_idx 0
[RX]   qos 1
[RX]   acm 0
[RX]   assoc_req_ie_len 78
[RX]   assoc_rsp_ie_len 98
[RX]   aid 0
[RX]   band 0
[RX]   center_freq 2437
[RX]   width 0
[RX]   center_freq1 2437
[RX]   center_freq2 0
[WF][SM] Exiting connecting state
[WF][SM] State Action ###connecting### --->>> ###wifiConnected_ipObtaining###
[WF][SM] Entering wifiConnected_ipObtaining state
[WF][SM] DHCP Starting...0x4201f73c
[WF][IF] Skip DHCPCing...
[APP] [EVT] connected 448269
[APP] [EVT] connected 448270
[17:20:44.801] - Info:MAIN:Time 447, idle 0/s, free 103912, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21
[...]
[17:20:58.850] - Info:MAIN:Time 461, idle 0/s, free 103912, MQTT 0(0), bWifi 0, secondsWithNoPing -1, socks 2/21
[17:20:59.351] - [WF][SM] IP obtaining timeout
Disconnect CMD
[RX] sm_disconnect_ind
       reason_code 0
[RX]    vif_idx 0
[RX]     ft_over_ds 0
sending disconnect
[WF][SM] Exiting wifiConnected_ipObtaining state
[WF][SM] State Action ###wifiConnected_ipObtaining### --->>> ###disconnect###
[WF][SM] Entering disconnect state
[WF][SM] Will retry in 2000 ticks
[APP] [EVT] disconnect 463271, Reason: Connection OK
[APP] [EVT] disconnect 463271
Info:MAIN:Main_OnWiFiStatusChange - WIFI_STA_DISCONNECTED - 2

openshwprojects, 2023-10-03:
hello, do you have a second device with the same MAC on your network?

Thor368, 2023-10-03:
Well, to be honest it looks like it tries to use the MAC 00:00:00:00:00:00 doesn't it?

openshwprojects, 2023-10-04:
I've seen similiar issue somewhere, but I don't know what causes it. I have tried to add MAC change functionality once but I just bricked my BL602 dev board that well, it seems that efuse is write-once feature or smth... have you cleared whole flash before flashing? That might have caused it. Do you have a backup?

Thor368, 2023-10-06:
I do have more of these LED controllers. So the OEM firmware would be obtainable but I have no clue how to do that with the "BLDevCube.exe" tool version 1.4.8. Same goes for clearing the whole flash. As a side node AP mode works fine. It even shows me a MAC address in the setup webpage.

openshwprojects, 2023-10-06:
I don't know much about BLDevCube but I remember we've tried it with another user and there is some option to read flash data if you select "Advanced" mode of that tool. Would you like to try? Of course, try to do flash reading first, I do not currently know how to recover from that MAC address issue...

howfastcanyouace, 2023-11-19:
I have same issue as op, figured out it has something to do with the networks security setting wpa2/wpa3. AP mode works fine and once I tested another router/network with security disabled the device connects without an issue.