openshwprojects/OpenBK7231T_App #936: Script's label not found

szupi-ipuzs, 2023-10-06:
**Describe the bug**
I believe that the interpreter is incorrect when saying:
```
Info:EVENT:EventHandlers_ProcessVariableChange_Integer: executing command if $CH4>$CH60 then "startScript autoexec.bat AUTO_TRIGGER_FULL"
Info:CMD:Label AUTO_TRIGGER_FULL" not found in autoexec.bat - will go to the start of file
Info:CMD:CMD_StartScript: started autoexec.bat at label AUTO_TRIGGER_FULL"
```

Here's the complete script, as you can see the label is there. Could it be the problem with order of labels in the file?
```
led_enableAll 0
alias schedule_light_off addRepeatingEventID 10 -1 123 led_enableAll 0

if $CH50==1 then "startScript autoexec.bat SET_AUTOMATIC"
if $CH50==0 then "startScript autoexec.bat SET_MANUAL"
if $CH51==0 then "SetChannel 100 3700"
if $CH51==1 then "SetChannel 100 3650"
if $CH51==2 then "SetChannel 100 3600"

AddChangeHandler Channel50 == 1 startScript autoexec.bat SET_AUTOMATIC
AddChangeHandler Channel50 == 0 startScript autoexec.bat SET_MANUAL
AddChangeHandler Channel51 == 0 SetChannel 60 3700
AddChangeHandler Channel51 == 1 SetChannel 60 3650
AddChangeHandler Channel51 == 2 SetChannel 60 3600
return

SET_AUTOMATIC:
led_enableAll 0
AddChangeHandler Channel3 == 1 if $CH4>$CH60 then "startScript autoexec.bat AUTO_TRIGGER_FULL"
AddChangeHandler Channel3 == 0 if $led_enableAll==1 then "startScript autoexec.bat AUTO_SCHEDULE_DIMMED"
return

SET_MANUAL:
cancelRepeatingEvent 123
cancelRepeatingEvent 321
led_enableAll 1
return

AUTO_TRIGGER_FULL:
led_enableAll 1
cancelRepeatingEvent 123
cancelRepeatingEvent 321
return

AUTO_SCHEDULE_DIMMED:
if $CH53!=0 then "backlog addRepeatingEventID 10 -1 123; startScript autoexec.bat AUTO_SCHEDULE_OFF" else "goto AUTO_SCHEDULE_OFF"
return

AUTO_SCHEDULE_OFF:
if $CH56!=0 then "backlog addRepeatingEventID 10 -1 321; startScript autoexec.bat AUTO_SET_OFF" else "startScript autoexec.bat AUTO_SET_OFF"
return

AUTO_SET_OFF:
led_enableAll 0
return

```

**Firmware:**
- Version 1.17.265
- Device LSC Sensor LED Flood Light
- Chip/model: BK7231T

**Device config:**
```
{
  "vendor": "LSC",
  "bDetailed": "0",
  "name": "LSC Sensor LED Flood Light",
  "model": "3003366",
  "chip": "BK7231T",
  "board": "TODO",
  "flags": "267264",
  "keywords": [
    "LSC",
    "Floodlight"
  ],
  "pins": {
    "6": "dInput;3",
    "8": "PWM;1",
    "9": "PWM;2",
    "23": "ADC;4"
  },
  "command": "",
  "image": "https://templates.blakadder.com/assets/device_images/lsc_smart_connect_3003366.webp",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}
```

szupi-ipuzs, 2023-10-06:
Hmm, I've changed the order but it didn't help, any ideas?

openshwprojects, 2023-10-07:
Hello, if you know how, create a self test for windows to replicate.

Or just try:

alias starter startScript autoexec.bat AUTO_TRIGGER_FULL
if $CH4>$CH60 then starter

szupi-ipuzs, 2023-10-14:
Hi, I restructured the script and eventually got it working somehow, here's the script now:
```
goto SETUP

SET_AUTOMATIC:
cancelRepeatingEvent 123
cancelRepeatingEvent 321
led_enableAll 0
return

SET_MANUAL:
cancelRepeatingEvent 123
cancelRepeatingEvent 321
return

ON_PIR_TRIGGERED:
if $CH51==0 then return
if $CH4<=$CH60 then return
led_dimmer 100
led_enableAll 1
cancelRepeatingEvent 123
cancelRepeatingEvent 321
return

ON_PIR_TRIGGER_OFF:
if $CH51==0 then return
if $led_enableAll==0 then return
AUTO_SCHEDULE_DIMMED:
if $CH54!=0 then "backlog addRepeatingEventID $CH54 -1 123 startScript autoexec.bat AUTO_SCHEDULE_OFF" else "goto AUTO_SCHEDULE_OFF"
return

AUTO_SCHEDULE_OFF:
led_dimmer 50
if $CH57!=0 then "backlog addRepeatingEventID $CH57 -1 321 startScript autoexec.bat AUTO_SET_OFF" else "goto AUTO_SET_OFF"
return

AUTO_SET_OFF:
led_enableAll 0
led_dimmer 100
cancelRepeatingEvent 123
cancelRepeatingEvent 321
return

SETUP:
SetChannel 51 $CH201
SetChannel 52 $CH202
SetChannel 53 $CH203
SetChannel 54 $CH204
SetChannel 55 $CH205
SetChannel 56 $CH206
SetChannel 57 $CH207
SetChannel 58 $CH208
SetChannel 59 $CH209

addEventHandler OnChannelChange 51 setChannel 201 $CH51
addEventHandler OnChannelChange 52 setChannel 202 $CH52
addEventHandler OnChannelChange 53 setChannel 203 $CH53
addEventHandler OnChannelChange 54 setChannel 204 $CH54
addEventHandler OnChannelChange 55 setChannel 205 $CH55
addEventHandler OnChannelChange 56 setChannel 206 $CH56
addEventHandler OnChannelChange 57 setChannel 207 $CH57
addEventHandler OnChannelChange 58 setChannel 208 $CH58
addEventHandler OnChannelChange 59 setChannel 209 $CH59

if $CH51==1 then "led_enableAll 0"
if $CH51==0 then "led_enableAll 1"
if $CH52==0 then "backlog SetChannel 60 3700"
if $CH52==1 then "backlog SetChannel 60 3650"
if $CH52==2 then "backlog SetChannel 60 3600"
AddChangeHandler Channel51 == 1 startScript autoexec.bat SET_AUTOMATIC
AddChangeHandler Channel51 == 0 startScript autoexec.bat SET_MANUAL
AddChangeHandler Channel52 == 0 SetChannel 60 3700
AddChangeHandler Channel52 == 1 SetChannel 60 3500
AddChangeHandler Channel52 == 2 SetChannel 60 3300
AddChangeHandler Channel3 == 1 startScript autoexec.bat ON_PIR_TRIGGERED
AddChangeHandler Channel3 == 0 startScript autoexec.bat ON_PIR_TRIGGER_OFF
```

So basically I wanted the lamp to have 2 modes:
1) manual, in which PIR is not used and the user can control all the parameters (on/off, dimmer, temperature)
2) automatic, in which PIR triggers light on with full brightness, but then after some (configurable) time (and when PIR detects no motion), the light should be dimmed to a (configurable) level and then after some (configurable) time the light should switch off.
And there's an **additional rule**: if the lamp is in automatic mode and the users changes either the lamp state (on/off) or temperature or brightness, then the lamp should immediately go to manual mode.

Here' the description of variables (ie. channels) I used (or plan to use):

```
---- SETTINGS (to be stored in flash) ----- 

MODE:
   CH51 = 0 -> Manual control (neither PIR nor timer control the light)
   CH51 = 1 -> Auto (PIR triggers light, then light is dimmed, then switched off)
   
AUTO_LIGHT_SENSITIVITY
   CH52 = 0 -> low (it has to be really dark for the algorithm to switch the light on)
   CH52 = 1 -> middle (it has to be moderately dark for the algorithm to switch the light on)
   CH52 = 2 -> high (it has to be a little dark for the algorithm to switch the light on)
   
AUTO_PIR_PROXIMITY
   CH53 = 0 -> near (PIR triggers light only when movement is very near)
   CH53 = 1 -> normal (PIR triggers light when movement is in moderate distant)
   CH53 = 2 -> far (PIR triggers light even on slight movement)
   
AUTO_LIGHT_FULL_TIME
   CH54 = X -> number of seconds of full light (can be 0, then this phase is skipped)
   
AUTO_LIGHT_FULL_DIMMER
   CH55 = X -> 0-100 
   
AUTO_LIGHT_FULL_TEMPERATURE
   CH56 = X -> 0-100 
   
AUTO_LIGHT_DIMMED_TIME
   CH57 = X -> number of seconds of dimmed light (can be 0, then this phase is skipped)
   
AUTO_LIGHT_DIMMED_DIMMER
   CH58 = X -> 0-100 
   
AUTO_LIGHT_DIMMED_TEMPERATURE
   CH59 = X -> 0-100 

---------- VARIABLES (to help with scripting) -------------------

AUTO_LIGHT_SENSITIVITY_ADC_VALUE
   CH60 = X actual adc value that CH4 should be compared against

```

The next step is trying to get it working in HA via MQTT.

However, the "additional rule" is problematic with mqtt, because I would have to prevent HomeAssistant from changing most of the variables directly, since I would have no way of detecting that change came from HA and not from the script itself (is there a way to detect this)?

Currently I'm toying with an idea of making HA use the command topic together with an alias (eg. "cmnd/OBK_DEV_NAME/cmd_turn_on"). This proves to work fine as long as the command does not require parameters, as there's no way to define parameters for aliases in the script (**is there**?).
Any other ideas?

openshwprojects, 2023-10-22:
Hello, I replied there:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20780008#20780008
let's talk on Elektroda for now