openshwprojects/OpenBK7231T_App #942: RGBIC chip Integration?

HalliHalloSchatz, 2023-10-19:
Is RGBIC possible?

What can I do, to support the integration of a RGBIC chip SM15155?

https://github.com/tuya-cloudcutter/tuya-cloudcutter/issues/503

openshwprojects, 2023-10-20:
This is a very interesting device indeed. I would love to support it!
Please consider posting a detailed teardown to our forum:
https://www.elektroda.com/rtvforum/topic3950844.html
Please provide information how this device worked with original Tuya app, were LEDs able to change color individually? How is the SM15155 connected to Beken? Which GPIOs are used? 
It is very possible that our SM16073 driver might even work for that.

HalliHalloSchatz, 2023-10-20:
How can I test the SM16073 driver? 
I found SM16073_DIN in GLIO-Finder? 
More documentation, what configuration I needed to get the sm16073 to work? 

Okay, I will try to write it down.

I the other "Issue" I wrote all informations, that I have at the moment.

Where can I measure the Pinout?
The PCb is very, small. Around 15mm.
There are only 8 golden pins.
I think, all other pins are to small for me to "Analyse".
There are only with a supermacro photoable. Only with eyes, there is nothing to see, so small are the PCB.

HalliHalloSchatz, 2023-10-20:
A simple voltmeter/ohmmeter helps...

Golden pins, from left to right

1: ground
2: 3,3v
3: to to PIN 16 BK7231N

Channel 16
I can set to low = ~0v
I can set to high = 3,3V

openshwprojects, 2023-10-20:
You dont need to set any pin roles. P16 is a hardware spi used LED control.

You just need to:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20719946#20719946

HalliHalloSchatz, 2023-10-20:
Wonderful!
I see the first light!

All 8 RGBCW LED are in parallel.
5CH = RGB ColdWihte and WarmWhite

In Log -> Command
1. Start driver
startDriver SM16703P
SM16703P_Init 1
SM16703P_SetPixel 1 255 0 0
SM16703P_Start
SM16703P_SetPixel 1 0 0 255
SM16703P_Start

Mhh, only green and no changes
How must I use this "Interface"?

HalliHalloSchatz, 2023-10-20:
I have add a Teardown in the Forum:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20778346#20778346

openshwprojects, 2024-04-05:
I will hopefully get soon this device. Let's continue discussion on forum.

openshwprojects, 2024-06-12:
Support added @HalliHalloSchatz , see: https://www.elektroda.com/rtvforum/viewtopic.php?p=21115906#21115906
I am closing this as solved