openshwprojects/OpenBK7231T_App #945: Incorrect color values from Hue Essentials / Tasmotrol

DiegoJp, 2023-10-21:
**Describe the bug**
When changing colors from one of these apps (Hue Essentials connected via [diyHue](https://github.com/diyhue/diyHue), or Tasmotrol) the RGB colors are set incorrectly.
For example when inputting the direct hex color from one of these apps:
[image]
the log is as follows in the webapp:

```
Error:CMD:BASECOLOR no sscanf hex result from ,0
Info:MQTT:Publishing val 0002000000 to obk42229CB7/led_finalcolor_rgbcw/get retain=0
Info:MQTT:Publishing val 38A221 to obk42229CB7/led_basecolor_rgb/get retain=0
Info:MQTT:Publishing val 15 to obk42229CB7/led_dimmer/get retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk42229CB7/led_finalcolor_rgbcw/get
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk42229CB7/led_basecolor_rgb/get
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic obk42229CB7/led_dimmer/get
```
Somehow the hex value #FF0000 gets converted to 38A221, and the light resulting it's a washed out green, which it must be a full red.

I've tried a combination of MQTT flags but to no avail, the above combination is 671155209 (flags 0 3 10 16 27 and 29)

I have no problems with Home Assistant, all colors are good are accurate (comparing hex values to channel values) regardless of any MQTT flag set in OpenBk. 

The same situation with others apps like Tasmotrol / HomeSwitch, etc

It should be said that others lamps configured in diyHue are OK and the colors are accurate as set.

In fact I have two OpenBk lamps (BK7231T and BK7231N) with the same behaviour (regardless on how many flags are set).

I do not know what's wrong (it may be diyHue or the apps) but I think these app may expect a Tasmota-like mqtt message but in the process the message gets converted to something and the resulting light is bad.

Please comment if you need more info/testing. To reproduce you can install Tasmotrol in an Android devices and test. You may also install diyHue in a docker container and add a Tasmota light to it (in the webapp the have) then installing Hue Essentials and changing the light there.

Regards, Diego Jp

**Firmware:**
- Version 1.17.271
- Device? Generic Tuya Lightbulb (no driver, direct PWM)
- Chip/model: BK7231T
- Device config? See attached images (Configure Module webpage):
[image]
[image]

openshwprojects, 2023-10-21:
hello, for some reasons scanf fails:
Error:CMD:BASECOLOR no sscanf hex result from ,0
but why? Can you capture somehow what's being send? Or should I add some more logging?

I can also try to get one of those apps and try

DiegoJp, 2023-10-21:
Hello, thanks for the quick reply. Capture how? with wireshark stuff? Perhaps i need to set a higher level of verbosity.

I'll be pleased if you trying these app on your own as my technical knowledge is limited.
Thanks is advance!

openshwprojects, 2023-10-21:
i 've foiund one error:
https://github.com/openshwprojects/OpenBK7231T_App/commit/bff7bb5d9e05a3f2855477d0e1a5b0776e6b6bb3
so far, one error, it caused sliders for hue, etc to "jump back to start" while changing color

DiegoJp, 2023-10-23:
> it caused sliders for hue, etc to "jump back to start" while changing color

Yes, exactly that. You're awesome the issue with the Tasmota apps was corrected, but now I'm having troubles with the diyHue pairing proces... please do not close this issue ye. I'll be updating if the problem it's here.

Many thanks!

DiegoJp, 2023-10-23:
nvm, it was an issue with my end. Thanks!