openshwprojects/OpenBK7231T_App #955: Home Assistant Discovery: voltage sensor gets overwritten by RSSI sensor

ntqbit, 2023-10-26:
**Describe the bug**
Home Assistant discovery doesn't discover voltage sensor. Instead, it is overwritten by RSSI sensor which gets the same "unique_id" due to the same index (see [rssi](https://github.com/openshwprojects/OpenBK7231T_App/blob/4d05eedc0cf87c4e911dc61a373a528f3e58c366/src/httpserver/http_fns.c#L2088), [power_source](https://github.com/openshwprojects/OpenBK7231T_App/blob/4d05eedc0cf87c4e911dc61a373a528f3e58c366/src/httpserver/hass.c#L408)).

In the `hass_populate_unique_id` function ([link](https://github.com/openshwprojects/OpenBK7231T_App/blob/4d05eedc0cf87c4e911dc61a373a528f3e58c366/src/httpserver/hass.c#L33)) both get the same unique_id: `OpenBK7231N_08C337A5_sensor_0`.

The contents of the topic `homeassistant/sensor/OpenBK7231N_08C337A5_sensor_0/config`:
```
{
	"dev": {
		"ids": [
			"OpenBK7231N_08C337A5"
		],
		"name": "obk08C337A5",
		"sw": "1.17.286",
		"mf": "Beken Corporation",
		"mdl": "BK7231N",
		"cu": "http://192.168.1.147/index"
	},
	"name": "RSSI",
	"~": "obk08C337A5",
	"avty_t": "~/connected",
	"uniq_id": "OpenBK7231N_08C337A5_sensor_0",
	"qos": 1,
	"dev_cla": "signal_strength",
	"stat_t": "~/rssi",
	"unit_of_meas": "dBm"
}
```

**Firmware:**
- Version: 1.17.291
- Chip/model: BK7231TN

**To Reproduce**
Steps to reproduce the behavior:
1. Configure MQTT
2. Configure Home Assistant discovery
3. Subscribe to `homeassistant/sensor/OpenBK7231N_08C337A5_sensor_0/config` topic.
4. Observe RSSI value instead of voltage.

openshwprojects, 2023-10-26:
Thanks, but, wasn't it fixed yesterday?
https://github.com/openshwprojects/OpenBK7231T_App/commit/1308a8f5b5ad226e4e155cbc0838c4e7733dcc81

ntqbit, 2023-10-26:
> Thanks, but, wasn't it fixed yesterday? [1308a8f](https://github.com/openshwprojects/OpenBK7231T_App/commit/1308a8f5b5ad226e4e155cbc0838c4e7733dcc81)

Seems like it.
Installed firmware a few days ago. I'll update now and check if it works.

openshwprojects, 2023-10-26:
I also have an idea for such a failsafe mechanism so it wont happen again:
[image]

ntqbit, 2023-10-26:
Yes. Everything works perfectly!
Thank you!

closed 2023-10-26 by ntqbit