openshwprojects/OpenBK7231T_App #963: Possible solution of config safe on W800

djcfg123, 2023-10-30:
I have the same problem as #913 . 
I accidentally chose 1.17.1 from the list and saving the configuration, parameters, etc. works on it, on the new ones it doesn’t work, immediately after the firmware the configuration file error and loading default. 
I have a suggestion: in the source code of the firmware offset 0x1F0303 is indicated, which is taken from the SDK example, that is, the recording area of ​​the configuration structure "mainConfig_t" which in earlier versions has the size MAGIC_CONFIG_SIZE_V3 2016 and works on the W800, and after adding SSID 2 has the size MAGIC_CONFIG_SIZE_V4 3584 and does not allow you to save the configuration on the W800. However, the user data area is at the starting address 0x8000000 + 1E0000, it might be worth shifting 1F0303 to 1E0000 (or so), unless of course there is a memory layout somewhere else that affects the beginning of the user block (maybe I didn't see this in the code).
Perhaps I would test this theory myself, but I’ve been trying to set up the SDK to compile the firmware for 2 days now and I’m getting build errors like “inttypes.h: No such file or directory”, “sys_rtos.h: No such file or directory” and similar

openshwprojects, 2023-10-31:
Hey, Why not use online builds? I can check this for you

openshwprojects, 2023-10-31:
https://github.com/openshwprojects/OpenBK7231T_App/pull/964

djcfg123, 2023-10-31:
Tell me, how can I use the online building? I tried to do this through the W800 SDK according to the guide written in BUILDING.md, but nothing works

openshwprojects, 2023-10-31:
I have merged that fix... for w800 config, dsoes it work now?

you can fork our repo and it should build on your fork, it will also build if you open a pull request after I approve your pull if you're a new contributor

djcfg123, 2023-10-31:
Yes, its working 😀
I was also able to fork and compile the firmware for w800, and there I shifted the address from 1F0303 to 1E0000 (as in the first message of this issue) and enabled SSID2 support for W800, it works, saves the configuration correctly, but I did not find the code itself  which uses the second network if it was not possible to connect to the first.
But is it still possible to enable IR support for the W800?

openshwprojects, 2023-10-31:
What has changed in RTOS? When? Can you point to specific commit that may break the system? It sounds like an important issue to fix.

openshwprojects, 2024-04-05:
Ok, no bug reports so far, so I think it can be closed.

closed 2024-04-05 by openshwprojects