openshwprojects/OpenBK7231T_App #967: Can't change startup command line in web server

Shestoperd, 2023-10-31:
**Describe the bug**
Using BK7231Flasher, I wrote down the firmware along with the configuration, went into the web server to change the startup command line, and was faced with the fact that I couldn’t do anything, at the end of the startup command line there was extra text, which apparently was part of the web form:`<input type='hidden' name='startup_cmd' value='1'><br><input type=`
Until I assembled everything into the case, I can change the settings from the BK7231Flasher, but after assembly it will be extremely difficult to do this.
I'm not sure that this is an error, perhaps the startup line is too long.
How to correctly change the startup command line through the web interface, it doesn’t even have a save button?

**Firmware:**
- Version 1.17.297
- Device:  HT01 Generic Temperature and Humidity Sensor  (HT01CB3S)
- Chip/model: BK7231N
- Device config `backlog startDriver tuyaMCU; startDriver tmSensor; linkTuyaMCUOutputToChannel 1 val 1; setChannelType 1 temperature_div10; linkTuyaMCUOutputToChannel 2 val 2; setChannelType 2 Humidity; linkTulinkTuyaMCUOutputToChannel 3 enum 3; setChannelType 3 ReadOnlyLowMidHigh; setChannelType 3 Battery;`

**Screenshots**
[image]

openshwprojects, 2023-10-31:
Hello, it looks like something in the command line broke the HTML of the page.
This is why save button is lost.
Can you try using at runtime command StartupCommand "test" ? this shouyld clear existing entry correctly

Shestoperd, 2023-11-01:
> Can you try using at runtime command StartupCommand "test" ? this shouyld clear existing entry correctly

Describe in a little more detail how to do this, just add the "test" command at the end so it turns out like this: `backlog Test; `?

Shestoperd, 2023-11-01:
It looks like it's a line length issue:
/src/httpserver/http_fns.c
```
void add_label_input(http_request_t* request, char* inputType, char* label, char* fieldId, const char* value, char* preContent) {
	if (strlen(preContent) > 0) {
		poststr(request, preContent);
	}

	hprintf255(request, "<label for=\"%s\">%s:</label><br>", fieldId, label);
	hprintf255(request, "<input type=\"%s\" id=\"%s\" name=\"%s\" value=\"%s\">", inputType, fieldId, fieldId, value);

```
If the resulting line is more than 256 characters, it is simply cut off and therefore the form breaks. It turns out that the length of the startup line should not exceed 208 characters.
How to reduce the length of the starting line without losing functionality?

openshwprojects, 2023-11-01:
Use autoexec.bat for long configurations.

openshwprojects, 2023-11-01:
I also added this fix for you:
https://github.com/openshwprojects/OpenBK7231T_App/commit/f9e5e428d0675ee2636dff09f1fa407d10916786

Shestoperd, 2023-11-01:
> Use autoexec.bat for long configurations.

Yes, I think it will be better this way.

> I also added this fix for you:

Thanks.

Shestoperd, 2023-11-01:
> I also added this fix for you:
> [f9e5e42](https://github.com/openshwprojects/OpenBK7231T_App/commit/f9e5e428d0675ee2636dff09f1fa407d10916786)

I think you can close it, everything works fine:
[image]

closed 2025-06-21 by openshwprojects