openshwprojects/OpenBK7231T_App #968: LittleFS config getting erased when updating to 1.17.299

Wolfieeewolf, 2023-11-01:
**Describe the bug**
Updated from Version 1.17.290 to 1.17.299.
LittleFS autoexec.bat config script was wiped. 
tried to restore fsblock but had to create a new file and start again.
happening on all devices

**Firmware:**
- Version 1.17.299
- Device? All - Switches, Lights, Plugs
- Chip/model: BK7231T, BK7231N. unsure if other chips are affected as I don't own any of them

**To Reproduce**
Steps to reproduce the behaviour:
1. Go to OTA update and update to version 1.17.299
2. Go to filesystem and list filesystem
3. See all scripts have been eraased

Wolfieeewolf, 2023-11-01:
So after some testing, the issue appears with 1.17.295.

It appears to only be affecting my BK7231T devices for now.

openshwprojects, 2023-11-01:
Was something changed in LittleFS back then? Or is it just caused by the size of OTA binary?

Wolfieeewolf, 2023-11-01:
The only things that appeared as changes were the buffer overflow and the PT6523 text (font). 

Doesn't look like anything with LittleFS was changed that I can see. My logs are not showing anything weird either. 

I thought it might be some issue with the OTA update not backing up the LittleFS. I did hit the readFS button and it said backing up. Not sure if that was it used for but it was worth a try. I then did the OTA update but it still erased it. Even when I hit the restoreFS it didn't do anyting

Whatever happened with build 1.17.295 has been carried over to all new builds. I have gone back to 1.17.290 and done the OTA with all new builds and it happens with each one.

Wolfieeewolf, 2023-11-01:
Just to add.

I can change the configs add new LittleFS scripts and restart the device and everything stays. I'm still working my way through some of the config/webapp changes but so far it appears to be only when you do an OTA update that it wipes the LittleFS.

openshwprojects, 2023-11-01:
THANK YOU, you are right, It was a silly #ifdef mistake! You can check now. It's nice that you caught it that early, thanks

Wolfieeewolf, 2023-11-01:
Perfect!! 

Just did the update to 1.17.301 and the LittleFS stays!!!

It's a pain when some small piece of code can make something not work.  

I'll close this out now that it has been solved.

closed 2023-11-01 by Wolfieeewolf