openshwprojects/OpenBK7231T_App #969: Flash ESPHome firmware OTA

lmcd, 2023-11-01:
I'd like to test something with ESPHome, but the generated firmware file isn't being accepted by the web interface:

`Invalid OTA file was dropped`

How do I overcome this?

avigeilpro, 2023-11-01:
OTA only accept openbeken firmware, more than this, it control that you put the good version of firmware (not a N firmware for a T controler for example). If you want flash another firmware i think the only way is by UART.

openshwprojects, 2023-11-01:
Unfortunatelly I do not provide any support for third party firmware, and especially not for the LT firmware, due to the repeated fraudulent behaviour of esphome port author towards the OBK team.

closed 2023-11-01 by openshwprojects

lmcd, 2023-11-01:
I managed to sort it in the end - it turns out ESPHome actually now generates openbeken firmware files that you can upload directly.

Regardless of any disputes with other projects, I think openbeken should allow you to flash with arbitrary firmware. I physically don't have access to UART and did everything with tuya-cloudcutter. In the spirit of openness, users should not be stranded with one specific kind of firmware, and should be able to migrate easily between various solutions.

I'm sorry you have a dispute with the LT people.
I'll say though, that beken support seems to be stabilising now across the ecosystem (in large part thanks to you 👏), and I think now might be the time to embrace the larger ecosystem/platforms and code interoperability. It will benefit everyone if everything works with PlatformIO etc, and allow openbeken to focus on a superior, Tasmota-style interface that ESPHome doesn't offer.

Thanks for all your hard work!

pini72, 2023-12-09:
> Unfortunatelly I do not provide any support for third party firmware, and especially not for the LT firmware, due to the repeated fraudulent behaviour of esphome port author towards the OBK team.

You know, that's exactly what tuya say about you

openshwprojects, 2023-12-10:
I do not provide any support for LT project due to openly fraudulent behaviour of it's author, including several misleading statements and info/contacts digging from me personally, after openly lying to me about "not going to make another port/firmware" just to get me to give him more info.  I consider this person actions unethical and therefore I openly condemn it.

openshwprojects, 2023-12-10:
Not to mention that, after I gave (unwillingly) LT's author information and contacts, he has even publicly stated that he has nothing to do with me, he figured everything himself and it's just a coincidence that both me and I are from Elektroda forum (even tough I gave him information and help on that forum), and he gave me no credit whatsoever.

openshwprojects, 2023-12-10:
both me and him * are from Elektroda forum and from Poland, which is, of course, according to him, a total coincidence....