openshwprojects/OpenBK7231T_App #970: Roadmap/Milestones/Issue Tracker

Wolfieeewolf, 2023-11-01:
I have noticed that this repository has a lot of open issues and is hard to keep up with what is being worked on. I was wondering if some kind of roadmap or milestones should be set up to better keep track of what is happening with this project.

It would be good to see the issues broken into 'to-do,' 'in progress,' or 'future builds.' It would also be great to have some kind of issue tracker to categorize, prioritize, and assign issues. This way, the issues can be split into bugs, feature requests, etc. I understand the labelling system works well for this. Would be good this getting used more

I understand this is probably a big undertaking. I'm happy to lend a hand in helping with labelling issues or setting up a visual workflow board. I don't have a lot of experience in this area but am willing to help out if that is something you wish to set up.

openshwprojects, 2023-11-05:
While this seems as good idea, and something we wanted to do with @DeDaMrAzR , I am not sure if I would be able to keep up to it. The main reason is that I just check issues and forum every day and try to do the first thing that seems important and doable. There is not much "background work" in progress, I just do things. 

There are other things that could help us a lot, including:
- help on docs, documentation, testing
- help with making sample scripts and stuff, to be posted on Elektroda
- help with more teardowns
- help with more self tests for Windows build and simulator
- help with testing with stuff like Tasmota control, etc, etc Homekit
- if you can code, then C implementation help would be great as well

Wolfieeewolf, 2023-11-08:
That's fair enough.  Working on the important and doable issues is probably the best way to go.  I still think having a road map of feature requests and plans for future builds would be a great way for people to know what you are working towards.  Even if it's not well maintained it will still be interesting information to have. Maybe I will see what I can come up with and I'll post it here. 

I have been helping out with making teardowns and testing the best I can.  Unfortunately, a lot of the hardware I have appears to not want to play nicely and I've been unsuccessful in dumping the firmware.  I'm sure as my knowledge grows I'll have success in the end.

openshwprojects, 2023-11-08:
Thanks!
Another great thing to do would be to go through forum topics and add missing devices to:
https://openbekeniot.github.io/webapp/devicesList.html
Namely, pull request them there:
https://github.com/OpenBekenIOT/webapp/blob/gh-pages/devices.json

Wolfieeewolf, 2023-11-19:
I have been playing around with making a website with Jeykll and would like to add a few things to the JSON template so it is easier to sort the devices.  

Can I ask what are you using to edit the devices.json file? Are you just adding new devices by hand or are you using a database manager or Excel spreadsheet?

openshwprojects, 2023-11-19:
That's great news, let me hear your suggestions. What would you like to add?
I am adding new JSON entries by hand, in notepad.

Wolfieeewolf, 2023-11-19:
I would like to add things like device type, category, device standard, model, etc. This should help sort the devices a little better. 

The website is still very rough and this is the first time I've used Jekyll. At the moment I'm just trying to set it up the same as what you already have and then tweak it from there. When I have something that works I'll let you know.

Wolfieeewolf, 2023-11-19:
Using https://jsoneditoronline.org/ I have managed to export the file as a CSV and do some editing.  With some tidying up it should be good to use but I will have to add new devices by hand as the new format most likely will not work with the OBK firmware. 

I've uploaded the CVS and the JSON file I'm using for testing if you are interested. Be aware this is not the final version so it looks a bit messy for now.

[devices2.csv](devices2.csv)
[devices2.json](devices2.json)