openshwprojects/OpenBK7231T_App #980: Is there a library that supports SPI?

ngoclaithe, 2023-11-08:
I'm working with CBU and SH4. Is there any library that supports reading and writing data from SH4?

openshwprojects, 2023-11-08:
I think that our contrbutor @DeDaMrAzR is also working on that device, we can speak about that on our forum. He is capturing packets for that, but I may be wrong.
We have SPI used for BL0942, so we can base our code on that.

ngoclaithe, 2023-12-12:
@openshwprojects I'm trying the SPI driver for SH4 based on BL0942, but I can't find information about the frame structure of sh4 sub-g 433m, there is none. Can you help me with information about the frame structure of sh4? Below is a picture of frame structure BL0942. 
[image]