openshwprojects/OpenBK7231T_App #981: [enhancement] Allow MQTT topic rewriting

analystcmyk, 2023-11-14:
I want to use OpenHASP https://www.openhasp.com/0.7.0/ to display sensor values published by OpenBEKEN to MQTT.

OpenHASP allows the values of screen objects to be set via MQTT.

For example : by publishing `25 `to `hasp/plate/command/p0b5.val` you update the displayed value of that object. 

### Issue : 

OpenBEKEN publishes to a fixed topic, eg `/device/power/get`

I cannot instruct OpenBEKEN to publish the sensor value to `hasp/plate/command/p0b5.val`

### Suggested :

Add command to rewrite the MQTT topic _just_ before publishing, e.g.

`mqttTopicRewrite  /device/power/get  hasp/plate/command/p0b5.val `

### Alternative :

use addEventHandler OnChannelChange 1 

Related issue : https://github.com/HASwitchPlate/openHASP/issues/577

openshwprojects, 2023-11-14:
Hmmmm arent you referring to this?
[image]
Third argument set to 1?

analystcmyk, 2023-11-14:
Ok, will check that first.

analystcmyk, 2023-11-14:
Ok yes this is working fine. 

`addEventHandler OnChannelChange 2 publish hasp/plate/command/p0b5.val $CH2 1`

openshwprojects, 2023-11-14:
btw, it seems you are doing something very interesting. Please consider writing a tutorial/;article about it for our forum, we can send you a set of Elektroda gadgets for that

analystcmyk, 2023-11-14:
Well before that  ( I am using the https://www.openhasp.com/0.7.0/hardware/sunton/esp32-2432s028/#esp32-3248s035 ) module I would like to elaborate a bit the back route from LCD module to OpenBEKEN. 

Unfortunately I have an issue with the touchscreen so I should wait for a new one, but generally speaking having OpenHASP to monitor/control OpenBEKEN is not a bad option. Just a dumb broker is needed to do all, no nodered or other dependencies. 

These modules have an ESP32 so that should be enough to run a micro MQTT server next to it. ( looking at https://github.com/tuanpmt/esp_mqtt ) 

But to be honest I would like be able to communicate touch events to openBeken directly, but since I have limited control on what topic openHASP is publishing, you could think of a flexible way to have OpenBEKEN actually subscribe to a user defined  topic. Maybe keep the (non-existing) command `mqttTopicRewrite`  analogy.

So instead of getting the commands from `OBK_DEV_NAME/INDEX/set` you would say : 

`mqttTopicRewrite OBK_DEV_NAME/1/set  Gadgetx/switch/1`

Of course that is not an easy thing to do, and it would need to honour the `CMD_SetChannelPrivate` as well.

Monitoring multiple OBK on a single OpenHASP already works fine.

openshwprojects, 2023-11-14:
So it would send that JSON to obk?

No problem, I can make a custom driver for that

analystcmyk, 2023-11-15:
A bit like this 

[image]

Angel0ffDeath, 2023-12-28:
You run usually MQTT server on home automation server. I don't know what home automation you are using, but it is always possible when receive a mqtt message in one topic just to change what you want (if necessary) and publish the message into another topic.
Your initial question is somehow MQTT server to receive the messages and in the same time your device, without server to do anything. Don't forget flash memory of the device is usually not so much and firmware fits barely. And firmware continues to grow, adding new devices, so it is good idea if your MQTT server can do some work instead of the device.
Usually the device needs just to report the data to server. What server will do with this data and how this data will be distributed to the other devices is server job and not device job.