openshwprojects/OpenBK7231T_App #983: New sensor CHT8310

Tobias0110, 2023-11-16:
I want to thank you all for the great firmware you created here. I successfully installed it on many devices.

Recently I came across a temperature and humidity sensor that is currently not supported (SensyLink CHT8310).
I tried to use it with the settings for the SensyLink CHT8305 but this results in broken measurement data, so it seems the data format is different. I can get a temperature reading that changes when heating / cooling the sensor but the value is negative at room temperature (-21 is displayed at about 22°C room temperature, moves towards 0 when heated). The humidity reading jumps randomly.

Sensor is connected to:
P20 [SCK]
P22 [SDA]

Modul in use: [tuya CBU](https://developer.tuya.com/en/docs/iot/cbu-module-datasheet?id=Ka07pykl5dk4u)

Datasheet is available on the [SensyLink Website](https://www.sensylink.com/html/web//product/wsdcgq/cwsdcgq/1663024476742897666.html).

[Bought on Aliexpress](https://de.aliexpress.com/item/1005005869874007.html?spm=a2g0o.order_list.order_list_main.5.12425c5fdgkmEW&gatewayAdapt=glo2deu).

[image]

[image]

[image]

openshwprojects, 2023-11-17:
The datasheet redirects me to login form , are you able to download it?

Tobias0110, 2023-11-17:
[CHT8310_Datasheet.pdf](https://github.com/openshwprojects/OpenBK7231T_App/files/13398333/CHT8310_Datasheet.pdf)

Yes, please see attached. You can create an account with random information. They don't require you to confirm the E-Mail address.

openshwprojects, 2023-11-18:
Is the datasheet only those 4 pages you attached?

Tobias0110, 2023-11-18:
Yes this is all for now. I am trying getting the full version containing more information about I2C.

openshwprojects, 2023-11-18:
Maybe just I2C address is wrong? Maybe we could try doing I2C scan?

Tobias0110, 2023-11-18:
If the address is wrong I would expect that we wouldn't receive any data. As descried in my initial post the received values correlate with the real temperature in some way.

I tried to contact the IC manufacturer. Lets wait a few working days for them to respond.

openshwprojects, 2023-11-18:
Well, are you sure? Change SDA and SCL to random pins and check what kind of data you receive. Hmmm

Tobias0110, 2023-11-18:
Nothing, all zero when using different pins...

Pinout is in the datasheet and was traced back from the sensor to the uC via a continuity tester.

Tobias0110, 2023-11-20:
SensyLink answered my E-Mails and was kind enough to provide the advanced datasheet.
They also included sample C code. I informed them what we are doing so it should be fine to post it here.

So far I didn't compare it to the CHT8305 docs.

[CHT8310 Advanced Datasheet_Ver1.0 20230407.pdf](https://github.com/openshwprojects/OpenBK7231T_App/files/13413436/CHT8310.Advanced.Datasheet_Ver1.0.20230407.pdf)

baustromverteiler, 2023-12-06:
So I also got this sensor from Aliexpress and tried my luck with modifying the sourcecode for the cht3805 but unfortunatley I am by no means a programmer ;) I think the difference in the two sensors is in the calculation of the values, the temp and hum from the cht3805 are calculated this way:

`uint CHT8305_get_temp_humi(float *t, float *h)

{

	CHT8305_read_reg(CHT8305_REG_TEMP, 4);

	unsigned int th, tl, hh, hl;

	th = buf[0];

	tl = buf[1];

	hh = buf[2];

	hl = buf[3];

	(*t) = (th << 8 | tl) * 165.0 / 65535.0 - 40.0;

	(*h) = (hh << 8 | hl) * 100.0 / 65535.0;

	return 1;

}`

(taken from: https://github.com/sensylink/CHT8305/blob/master/CHT8305.cpp)

while the temp and hum for the cht8310 is calculated like this (sorry for the screenshot, the PDF didnt allow to copy):

[image]

(taken from the datasheet provided by op)

I tried to modify the calculation part with the values taken from the cht8310 datasheet but I failed to to get correct readings, maybe this information helps somebody cleverer than me to figure out how to get this sensor working

divadiow, 2023-12-17:
Hello. Just wondering if any progress has been made on a modified driver for this sensor. I posted about it here https://www.elektroda.com/rtvforum/topic4019974.html

artin961, 2023-12-22:
Hello, Don't know if someone else is working on the popic but I happen to have this sensor and sarted thinkering with it.  

The sensor is reading the temperature and humidity, but humidity is a bit jumpy. I even managed to utilise the low power feature of the sensor where it is not measuring all the time and only on command. As a result it should consume onlu 50nA of quiescent current. 

I made the changes in the current driver for 8305, using 0xff register that should contain 0x8305 on the original sensor, but on this sensor there is left undocumented. My implementation assumes it is 8310 if cannot identify it as 8305. But this needs to be verifyed by someone with 8305 that is still functioning correctly. 

Anyone interested could check [https://github.com/artin961/OpenBK7231T_App/tree/add_CHT8310_support](url) and please give feedback.

artin961, 2023-12-23:
Reading from the sensor looks good on the scope but the humidity is jumping all over the place. 
[image]
Temperature reading is stable and reliable, havent figured out how to sleep and wake it on timer so I'm runing the sensor on a lab power supply. 
[image]

artin961, 2023-12-23:
I managed to get it working. On this sensor humidity data cannot be read from 0x00. When you read 4 bytes from there the third one is parity information and last 0xff. If it is interpreted as humidity it is all over the place randomly as parity changes whenever temperature bits change. I left it to be that way to not break compatibility with the old sensor, but overwriting the last 2 bytes from direct read from the second register where humidity data is 0x01.

artin961, 2023-12-24:
Tested the sesnor runing on batteries with deepsleep, waking every 15 minutes to measure. It appears to be stable. 

[image]
[image]

closed 2023-12-24 by openshwprojects

openshwprojects, 2023-12-24:
Related testing thread: https://www.elektroda.com/rtvforum/topic4019974.html

openshwprojects, 2023-12-24:
@artin961 good job! Thank you

NonPIayerCharacter, 2023-12-31:
@artin961 On CHT8310, when room temperature is below zero, sensor temperature readings looked nonsensical (~250 celsius), as i understood, the real value is (temperature - 256).
[image]