openshwprojects/OpenBK7231T_App #985: New Wifi Presence Sensor - ZY-M100

Hollako, 2023-11-24:
Hello,

I have bought a new presence sensor Ceiling mount the ZY-M100 Wifi version. I opened it and found a WBR3 chip inside however I couldn't find it in the compatible OpenBK products database.

[image]

Hence, in order to contribute in this project, I am glad to help by flashing this device with OpenBK firmware then make the necessary tests and confirm its fully functionality with OpenBK firmware then add it to the list of compatible product database.

The device has a Tuya MCU,  my question here after flashing OpenBK firmware, what is the next step ?

Thank you in advance.

CBDesignS, 2023-11-24:
I would hazard a guess that if you can see it on wifi and log into it you can then see if you can start setting up pins etc. did you backup the original firmware first, if so you can use the tools to see if you can get an idea of the original config setup and take it from there.. I am still llearing my way about so not the best to give advice. there should be guides to get you started if you check out the flasaher app.

Hollako, 2023-11-24:
Hey @CBDesignS ,

Actually, I made it and now the sensor is working with OpenBK firmware however it was not straightforward with the Chip WBR3. Here are the steps I did:

1- Replace the WBR3 Chip with the CB3S since I couldn't flash the WBR3 (am not sure if it was my fault or it's not supported yet)
2- Luckily I found in the Tasmota document the dpID of the TuyaMCU and I implemented them in the OpenBK firmware I am listing below the autoexe.bat command that I used.

**startDriver TuyaMCU
tuyaMcu_setBaudRate 115200

setChannelType 1 OpenClosed
setChannelLabel 1 "Presence"
linkTuyaMCUOutputToChannel 1 bool 1

setChannelType 2 Illuminance
setChannelLabel 2 ""
linkTuyaMCUOutputToChannel 104 val 2

setChannelType 3 ReadOnly
setChannelLabel 3 "Target Distance (cm)"
linkTuyaMCUOutputToChannel 9 val 3

setChannelType 4 TextField
setChannelLabel 4 "Sensitivty (0-9)"
linkTuyaMCUOutputToChannel 2 val 4

setChannelType 5 TextField
setChannelLabel 5 "Near Detection (0 .. 1000, 1 = 0.01m)"
linkTuyaMCUOutputToChannel 3 val 5

setChannelType 6 TextField
setChannelLabel 6 "Far Detection (0 .. 1000, 1 = 0.01m)"
linkTuyaMCUOutputToChannel 4 val 6

setChannelType 7 TextField
setChannelLabel 7 "Detection Delay (0 .. 100, 1 = 0.10s)"
linkTuyaMCUOutputToChannel 101 val 7

setChannelType 8 TextField
setChannelLabel 8 "Fading Time (20 .. 1500, 1 = 1s)"
linkTuyaMCUOutputToChannel 102 val 8

setChannelType 9 ReadOnly
setChannelLabel 9 "Self Check 0=checking, 1=check_success, 2=check_failure, 3=others, 4=comm_fault, 5=radar_fault, Current"
linkTuyaMCUOutputToChannel 6 val 9

setChannelType 10 ReadOnly
setChannelLabel 10 "Reset"
linkTuyaMCUOutputToChannel 106 bool 10**

Now the sensor as I mentioned before is up and running but I still facing some problems with the reliability and the parameters that are not reflecting exactly on the TuyaMCU, noting that the communication is working between the CB3S and TuyaMCU and I am able to get the Presence sensor status (Open-Close), illuminance and Target Distance. My problem is strictly with the Presence Sensor status and the fading time where I am not able to get how the Fading time parameter is reflecting on the behavior of the sensor. As per Tasmota document, they said that the parameter is in Seconds but I couldn't see this in OpenBK. For example, I put 60 seconds then the Sensor goes to Open after 13s or more or less.

openshwprojects, 2023-11-25:
This is a very good progress, but the place for that is our forum, if you can post full description there we can send you a set of gadgets:
https://www.elektroda.com/
The generic config looks okay and it seems you've done research very well, so it would be a great addition to our https://openbekeniot.github.io/webapp/devicesList.html !

Some hints:
- you can use sendstate command to send raw value to check how each value works
- if you want to make better ui, you can consider https://www.elektroda.com/rtvforum/topic3971355.html