openshwprojects/OpenBK7231T_App #986: BK7231T: slowdown and crash on scene switch with device groups

jcam, 2023-11-26:
**Describe the bug**
Scene switch stops responding when configured to both send and receive device groups commands

**Firmware:**
- Version: 1.17.291
- Device: CREE connected max scene dim controller
- Chip/model: BK7231T
- Device config:
```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Full Device Name Here",
  "model": "enter short model name here",
  "chip": "BK7231N",
  "board": "TODO",
  "flags": "671089664",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "0": "BP5758D_DAT;0",
    "1": "BP5758D_CLK;0",
    "6": "Btn_ScriptOnly;0",
    "7": "Btn_ScriptOnly;0",
    "8": "Btn_ScriptOnly;0",
    "14": "LED_n;6",
    "15": "AlwaysLow;0",
    "17": "AlwaysLow;0",
    "26": "Btn_ScriptOnly;0"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Configure switch to both send and receive all device groups types, with a named device group (_Jesse_Counter_)
2. Configure some lights in the same device group to receive only to that same group
3. Configure another set of lights on the same network, with a different device group (_Jesse_Bed_)
4. Issue a large number of DG control commands from the _Jesse_Bed_ group
5. Attempt to use the scene switch or log into it
6. Observe: Switch gets slower and slower and eventually stops responding either to physical input or to remote commands from MQTT/Home Assistant or the web client. There appears to be a memory leak in the receive code around receiving DG messages for a different group, as the same switch operated without issue for days before the second device group was added to the network, and also did not exhibit the problem after turning off power to that other group of lights.

I am not sure if the problem will occur when the scene switch is only configured for Send instead of both Send and Receive, but I have attempted changing it to that mode and will see if it exhibits the same behavior. I do note that it already is showing: **_DGR received: 23, send: 0_** despite not being configured to receive anything, and despite the other DGR group name not matching... i am not sure if it should be showing all messages on the network even if they do not match?

openshwprojects, 2023-11-26:
Hello, thanks for the detailed report. Before I have time to check this out, can you save your configuration externally and try some old version of OBK? I remember that I have tested DGR extensively few months ago or so and it seemed well stable. 

I also see that you suspect the memory leak - does the heap size in every second print go down?

Does the problem allso appear if you use... a smaller number fof DG control commands? I am reffering to the point 4. I want to know whether the problem is related to inability to handle large number of packets, or is it a general issue.

g_dgr_stat_received  gets increased everytime anything is received, the group name filtering happens later:
https://github.com/openshwprojects/OpenBK7231T_App/blob/801959d9b8252d9c65ef48f08e3d6fe0d397e78a/src/driver/drv_tasmotaDeviceGroups.c#L627

Still, I think that the receive function could be optimized right now, but I don't know if it would help.
I can optimize for example this:

		myip = HAL_GetMyIPString();
		me.sin_addr.s_addr = inet_addr(myip);
		
This is unnecesssarily done every quick tick right now.... it wasn't the issue before, but maybe it's for you? I can try to fix it today or so, but I will also have to update binaries for you manually because there is some miscellaneous issue with Github builds.

openshwprojects, 2023-11-26:
Whoa, there is a lot to opti9mize in that function, let me get to it, can you help me with testing? I also would like to make sure that I will nto break anything!

jcam, 2023-11-26:
I'd be happy to test it out (though I can't necessarily commit to fast turnaround as I have some family obligations later today), though I probably will need some guidance on what specific things to look out for with testing. 

I just tried with the webapp logs running on the Jesse_Counter scene switch, I attempted sending dozens of commands with the Jesse_Bed group and I don't see the 'free' counter dropping at all.

One thing I am not sure of is how long it takes for the switch to lock up, it could be unrelated to the commands being sent perhaps? I am not sure... I will try just leaving things set up as it had been before but with the logger running, and hopefully I can catch something that would indicate what it is doing.

openshwprojects, 2023-11-26:
What do you mean by "for the switch to lock up"? I am not sure if I understand.

I also noticed that the debug log output calls the IP to string conversion EVERY time, it's yet another mistake of mine. I will fix it in next commit, but I will first check it on my side as well.
[image]

The little issue is that we have BK7231T building problem on Github builds, so I will attach RBL here as an attachment....

There is no hurry with testing, I will do some improvements on my side and then wait for your turn.

Just keep in mind that I will need you to test two things:
1. whether my changes broke something or not - you know, it happens... so keep an eye whether everything works at least as good as before
2. whether your device stability has improved...

I don't have that large setup as you so OBK may behave differently in your environment.

jcam, 2023-11-26:
Yes, the stability problem seemed to take anywhere from a couple of hours to a day before it became obvious something was failing. 

My setup currently is that I have 11 bulbs and one switch in my apartment, the switch and 4 bulbs near the counter are one device group, then there are 4 bulbs near the bed that are all another device group, and the remaining 3 bulbs are all running individually with DGR turned off.

I was thinking the problem is in DGR because when the switch locked up completely I had to power cycle the circuit breaker feeding it... in the process the power supply for the 4 bulbs by the bed gets turned off. I turned it back on, and then maybe half a day later I noticed the switch had stopped responding again, so I reset the circuit breaker another time... but this time I forgot to turn the bed lights' power supply back on, so they stayed turned off for a day... everything was stable, then I turned the bed lights back on and I noticed that within 2 hours the switch had started getting slow again.

I had tried using home assistant to change the color/brightness/etc of the Jesse_Bed group, and as mentioned in my last comment the free counter didn't go down. Well, I just tried changing colors, brightness, etc through home assistant of the **_Jesse_Counter_Switch_** while watching the log, and I noticed that the free counter did drop, by maybe 150 bytes, then by that amount again. Nothing specific caught my eye in the log. Unfortunately the drop had scrolled off the screen before I copied it down, and of course _now I can't get it to happen again_...

jcam, 2023-11-26:
Since I don't have specific reproduction steps that make it happen every time I can't guarantee what is the cause or whether a new build will fix it, but yes if you have a manual build that you know at least won't brick my devices, I would be fine with loading it up from here, I don't mind that it isn't coming from a github actions build.

openshwprojects, 2023-11-26:
I can never know anything for sure, but at least I can assure you that I test my builds on my devices, so the chance of bricking should be really, really low.

I will try to do some minor yet substantial fixes and we will see if stability on your side improves.

|Btw, have you tried to enable extradebug? The increased log level? To see what is printed?
[image]
Still, it may not be the best idea because extra logging actually slows down everything a lot...

jcam, 2023-12-07:
just finally getting back to this, apologies. i'll load this up on my switch today!

jcam, 2023-12-07:
Okay I've loaded the 1.17.335 build onto the switch and also 10 of the lights (2 DGR groups of 4, one of those with the switch, 2 individual).
So far basic functionality is good, all lights switch and change colors as they should.
As a followup from before, i did wait with the old build until it was starting to get super slow... the memory free number had *not* gone down, so it wasn't a memory leak, perhaps? if it happens again i'll try flipping on debug logging and see if it is "doing something" under the hood on repeat

jcam, 2023-12-19:
1.17.335 has been working without issue as far as I've observed for the last 2 weeks! I think I am going to call this fixed :)

I'm going to say resolved by: 3cd75b92e9bb0e88a92cd17e2cbbc23781dfa6ce

closed 2023-12-19 by jcam