openshwprojects/OpenBK7231T_App #988: 2 separate PWM channel to Home assistant

TomaszCwik, 2023-12-02:
Hello,
I have problem with connect two separete PWM channel to Home assistant
1. I use magic home on Chipset BL602 and openbk in  version 1.17.333
2. I Use 2/3 pinout to connect separete Led Strip (White only)
3. after configure 2 chanels as PWM OpenBK and HA see my led stript as RGB strip or Warm/Cold Strip
4. after run Flag 3 i proper see separate channels in OpenBK but HA still discover my pwm as warm/cold strip
How should i proper config separete channels in openbk ?

openshwprojects, 2023-12-02:
Hello, just add the device with YAML code directly. Or... maybe I can look into it alter and fix HASS discovery to respect that flag.

TomaszCwik, 2023-12-02:
mayby its becouse the flag is mark as debug ?
[image]
after turn that flag on i don't see toogle buttons but i can controll channels using bar
[image]

TomaszCwik, 2023-12-02:
should i made change in this yaml ?
[image]

4pack, 2024-02-12:
This would be helpful on my end as well. I asked essentially this question #896 a year ago, and Home Assistant did pick up both of them, although with some weirdness.