openshwprojects/OpenBK7231T_App #991: Safe Mode not work, Secondary Wifi not use.

exotsk, 2023-12-04:
**Describe the bug**
A clear and concise description of what the bug is.

**Firmware:**
- Version OpenBL602_1.17.333.bin
- Device? 
- Chip/model: BL602

After start setting i have this error. i was set second wifi which module not use.

Дек 4 22:20:02 ndm
Network::Interface::Mtk::WifiMonitor: "WifiMaster0/AccessPoint0": STA(ac:d8:29:06:8b:3d) pairwise key handshaking timeout (msg 1 of 4-way).
Дек 4 22:20:02 ndm
Network::Interface::Mtk::WifiMonitor: "WifiMaster0/AccessPoint0": STA(ac:d8:29:06:8b:3d) had deauthenticated by AP (reason: PTK 4-way handshake timeout).

i can`t run safe mode by switching module less than a 30 sec many times.
it`s always try to connect to first wifi. (pasword mistake or mesh mode, i dont know yet)

openshwprojects, 2023-12-04:
Secondary WiFi is not  fully implemented, I apologize for the problem. The field on GUI is a placeholder. We need to finish code for that.

exotsk, 2023-12-04:
thak you for answer. but power on-off did not work too. I had to solder again and rewrite the module in the erase option.
any attempts to switch to safe mode up to 20 times were not successful. it's not clear what I wasn't doing right.

openshwprojects, 2023-12-04:
have you waited long enough for capacitors to discharge? do you have a failed boots couinter on web panel?

exotsk, 2023-12-05:
I did not focus on the shutdown time, I think about a second, but I tried to keep on for more than 3 seconds, and of course not more than 30

rafalolb, 2024-02-14:
Is available fix for configure secound wifi?

xjikka, 2024-05-27:
HI, here is my SSID2 solution : https://github.com/openshwprojects/OpenBK7231T_App/issues/1246