openshwprojects/OpenBK7231T_App #995: OTA to esphome

espilioto, 2023-12-11:
I'm trying to flash an rbl compiled form esphome to test some stuff, but the frontend says _Invalid OTA file was selected_.
This is weird because a few days ego the exact same file was successfully flashed.

Tried on latest 1.17.335 and tried downgrading to 1.17.230 (the initial cloudcutter flash I had).

Any ideas?

openshwprojects, 2023-12-11:
I'm sorry but I do not provide support for third party firmwares, especially not for LT/Esphome due to repeated fraudulent behaviour of it's author.

closed 2023-12-11 by openshwprojects

openshwprojects, 2023-12-11:
I've done nothing to lock you in or screw you, I just check first 10 or so bytes of RBL header, which should match this pattern:
[image]
I don't even know what esphome guy puts there, maybe something else. Maybe if you change their header to match mine it may work better or not?

And yes, LT's guy behaviour is unacceptable. As far as I know, they have even invented their own OTA format, something UF2, which made OTA from OBK very hard for a long time, and it's of course just one of many other violations. Do you know that esphome port author reached me in the past personally just to dig some information, while saying that he has no intention of making BK or other firmware? And yes, then he used this information to make bk firmare.... not to mention, of course, that he's from same same forum as me (Elektroda), from the same country, etc, but he didn't even credit me, and even publicly stated it's a total coindicence that authors of two BK firmwares/ports are from the same country....

espilioto, 2023-12-11:
Well, (of course without having heard the other side) that does sound shady af.
Open source projects are supposed to be the exact opposite of that.

Nonetheless, thanks for your contributions and your effort.
They are greatly appreciated.