openshwprojects/OpenBK7231T_App #998: LSC 2578539 E27 RGBCW 806 Lumen new mapping ?

snakesrules94, 2023-12-13:
Hi there, as i have no account on Elektroda , i'll post here , feel free to copy/integrate the infos if needed .
I bought last week some E27 LSC rgb bulbs in Action store ( in France ) and it seems that the hardware config has changed .
Indeed , i can see in the profile/teardown that it was fitted with an BK231T [https://www.elektroda.com/rtvforum/topic3953325.html#20411879](url)
but in my case , there is an BK231**N** inside and it was with 1.3.21 FW.
The bulb was correctly flashed with TCC using the 1.3.21 profile .
After that and some trials/errors test, the correct config for this version is:

`"pins": {
    "7": "BP5758D_CLK;1",
    "8": "BP5758D_DAT;0"
  },
  "command": "BP5758D_Map 4 3 1 0 2"`

After that , all the controls are ok .

[image]

[image]

[image]

[image]

Unfortunatly, after a full dismount/remount of the first bulb , it's not working anymore , i think i have made a short circuit with the added capacitor soldered on the 6 pins connector of the leds circuit. I've desoldered the CB2L from the board but it seems that i can't flash it anymore , no answer by turning power on/off during flash attempt or doing the CEN reset . That was for the science :-)

openshwprojects, 2023-12-13:
Why have you added capacitor there?

snakesrules94, 2023-12-13:
Well , that's not me , it was factory mounted on the board ...

openshwprojects, 2023-12-13:
They surely chosen a strange way to mount a capacitor. Still, thanks, I will add an entry for that to the devices list.