openshwprojects/OpenBK7231T_App #999: Webapp disallowing cross-flashing for RBL files

Cossid, 2023-12-13:
**Describe the bug**
Cross-referencing https://github.com/OpenBekenIOT/webapp/issues/65 as I'm not sure how actively that is monitored being a separate org/project.

The check for rbl added in https://github.com/OpenBekenIOT/webapp/commit/e88762901e02d4b522afdebb951afb0162c4e5ae is preventing cross flashing, and would also prevent the requested feature https://github.com/openshwprojects/OpenBK7231T_App/issues/840

As mentioned in the comments, if simply checking for the start of RBL at 0-2 is not sufficient, a better check would be for app at 12-14

**Firmware:**
N/A - Web based issue.

**To Reproduce**
Steps to reproduce the behavior:
1. Attempt to flash non-OpenBK7231T_App RBL from web OTA interface

openshwprojects, 2023-12-13:
Thanks for pointing that out. It seems that I accidentally pushed test code to public and you are the first one to notice this. 
This will be fixed soon, I am going to introduce RBL strings at that place for N and T along with OBK build version and maybe some extra version check with a warning because downgrading may cause a config loss (in the past config structure was monolytic and it was not possible to add fields to it without breaking CRC).

Cossid, 2023-12-13:
Yes, a hard-block was the intent behind #840 when mixing up chip types (CloudCutter already enforces this when present, which is why I made that request).

I would encourage the version # check to only be a bypassable warning, as version numbers won't make sense for cross-flashing, and users won't be able to easily control the data there (unlike the filename).

Cossid, 2023-12-19:
I understand you want to improve this, but in the mean time, can you please remove the invalid `-o` check (which is in no way a valid RBL check)?  According to internet searches, I was not the first to report the issue, but support requests have been hostilely responded to and deleted, which I find unfortunate.

openshwprojects, 2023-12-20:
Please show me where are the hostile responses and I will ban the offender immediatelly. I am looking into the issue more right now.

Cossid, 2023-12-20:
The hostile responses appear to be from you stating you won't support 3rd party firmwares, when that isn't what users were requesting.  They were simply requesting not to be blocked from switching themselves.  Here is a discussion thread from reddit of at least two other users creating issue tickets and them being deleted, and one quotes a developer who appears to be you.

https://old.reddit.com/r/Esphome/comments/13d1q8g/openbeken_to_esphome_and_maybe_back_again/kc8hxlc/  Only two people attempted to create issues, but sever others have been having the same issue.

openshwprojects, 2023-12-20:
I apologize for the misunderstanding. By a "don't support LT" I meant that I don't check my firmware with their and I am not aware about their ecosystem. I totally ignore them now due to the repeated and deliberate fraudulent behaviour of their main author, Kuba2k. 

I am very suprised that simple unfinished version check is so problematic for people, last time I checked LT required totally different binary file and users had to use third party tool converter to migrate to OBK....

Still, I removed the check. I will also remove this issue because I do not want to promote the work of the fraudulent developer in any means necessary.

Cossid, 2023-12-20:
> I am very suprised that simple unfinished version check is so problematic for people, last time I checked LT required totally different binary file and users had to use third party tool converter to migrate to OBK....

The issue is that the tool built into OpenBeken was blocking them from doing it.  Yes, ESPHome requires specific files, but they also include [full instructions](https://docs.libretiny.eu/docs/flashing/esphome/#migrating-from-esphome-to-openbeken) on how to do it, and do not block anything.

This is all about users having choice of their own destiny.  There is no need to remove the issue, closing it and leaving it of public record is the best policy.

closed 2023-12-20 by Cossid

openshwprojects, 2023-12-20:
You are incorrect, no built-in tools were ever blocking anything. The unfinished check was in open source 3rd party web panel, that is outside the firmware.

The public record is a good idea when it applies to everybody, but as far as I know there are no public records of LT backstory and Kuba2k2 behaviour, which I consider unfair and harmful for OBK.

openshwprojects, 2023-12-20:
PS: I don't remember you saying to Digiblur that leaving posts public is a better policy.