openshwprojects/OpenBK7231T_App #1000: BK7231N Mini WiFi Smart Switch using TglChanOnTgl switches light on after restart if input pin active

xjikka, 2023-12-17:
On the BK7231N Mini WiFi Smart Switch, the correct state of the ADC input pin 
can be readed 1000us after the pin is initialized. Maybe there is a capacitor?
Without delay, g_lastValidState is after restart set to 0, so the light will toggle, if the switch on input pin is on (1).
So I added delay for 2000us and everything working good.

**Firmware:**
- Version [e.g. 1.17.360]
- Device BK7231N Mini WiFi Smart Switch,
- Chip/model: BK7231N
- Device config
  "chip": "BK7231N",
  "pins": {
    "0": "Rel;1",
    "8": "WifiLED;0",
    "10": "Btn_Tgl_All;0",
    "23": "TglChanOnTgl;1",
    "26": "Rel;1"
  },

**To Reproduce**
Steps to reproduce the behavior:
1. Activate switch so, that input is 1
2. Click on 'restart
3. Light is toggled

Solution:
I added delay for 2000us and everything working good.

////////////////////////////////////////////////////////////////////////////////////////
What I Changed:

@new_pins.c in PIN_SetPinRoleForPinIndex

immediatelly after these lines:
	HAL_PIN_Setup_Input_Pullup(index);
	// otherwise we get a toggle on start
	
added this:
	#ifdef PLATFORM_BEKEN
	//20231217 XJIKKA
	//On the BK7231N Mini WiFi Smart Switch, the correct state of the ADC input pin 
	//can be readed 1000us after the pin is initialized. Maybe there is a capacitor?
	//Without delay, g_lastValidState is after restart set to 0, so the light will toggle, if the switch on input pin is on (1).
	//To be sure, we will wait for 2000us.				
	usleep(2000);
	#endif

////////////////////////////////////////////////////////////////////////////////////////	
How I tested it:
@new_pins.c in PIN_SetPinRoleForPinIndex

#immediatelly after these lines:
	HAL_PIN_Setup_Input_Pullup(index);
	// otherwise we get a toggle on start

added this:
	int fpv[20];
	for (int fct = 0; fct < 20; fct++) {
	  fpv[fct] = PIN_ReadDigitalInputValue_WithInversionIncluded(index);
	  usleep(100);
	}
	for (int fct = 0; fct < 20; fct++) {
	  ADDLOGF_INFO("Pin %d after %d us state %d", index, fct * 100, fpv[fct]);
	}

output without delay:

Info:PINS:Pin 23 after 0 us state 0
Info:PINS:Pin 23 after 100 us state 0
Info:PINS:Pin 23 after 200 us state 0
Info:PINS:Pin 23 after 300 us state 0
Info:PINS:Pin 23 after 400 us state 0
Info:PINS:Pin 23 after 500 us state 0
Info:PINS:Pin 23 after 600 us state 0
Info:PINS:Pin 23 after 700 us state 0
Info:PINS:Pin 23 after 800 us state 0
Info:PINS:Pin 23 after 900 us state 0
Info:PINS:Pin 23 after 1000 us state 0
Info:PINS:Pin 23 after 1100 us state 1   - after 1ms +/- 100us  is state of pin correct
Info:PINS:Pin 23 after 1200 us state 1
Info:PINS:Pin 23 after 1300 us state 1
Info:PINS:Pin 23 after 1400 us state 1
Info:PINS:Pin 23 after 1500 us state 1
Info:PINS:Pin 23 after 1600 us state 1
Info:PINS:Pin 23 after 1700 us state 1
Info:PINS:Pin 23 after 1800 us state 1
Info:PINS:Pin 23 after 1900 us state 1
[new_pins.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/13696424/new_pins.zip)

openshwprojects, 2023-12-19:
That's interesting fix, is it only for ADC pin, or is it something more common? SHould we include it as a flag?

xjikka, 2023-12-19:
> That's interesting fix, is it only for ADC pin, or is it something more common? SHould we include it as a flag?

Because it is device from AliExpress (https://www.aliexpress.com/item/1005005351991955.html) and BK7312 become very common in these devices, I would definitely suggest to deal with it. However, because is it very specific (Belken Chip, the TglChanOnTgl mod and it is just about short 2ms delay), I would probably put a simple delay of 2ms for Belken and for TglChanOnTgl mode for now.

Using new flag would not address the eventual need to change the delay. Then I would probably solve it via the pin parameter in the config e.g. pins": { "23": "TglChanOnTgl;1;2"... (value 2 for the delay in ms) or something like that.

But as I already wrote, for now I would just put a delay there. Note that the delay must be between HAL_PIN_Setup_Input_Pullup and PIN_ReadDigitalInputValue_WithInversionIncluded.

I don't think it applies to the ADC pin. It depends on the circuit wiring. Dynamic of the ADC pin is just more dependent on the input circuit. In theory, this can apply to any pin on other platforms as well.

gutek18a, 2024-02-01:
Hi, could you please share the compiled version of the firmware with your patch?
I also have BK7231N Mini Switches that have the same problem, and I see that your changes have not been implemented in the main release.
Thank you.

xjikka, 2024-02-01:
@gutek18a Sure, here it is /rbl for web App OTA update/
[OpenBK7231N_dev_20231217_145714-rbl-webapp-ota.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/14130765/OpenBK7231N_dev_20231217_145714-rbl-webapp-ota.zip)

gutek18a, 2024-02-02:
Thank you very much!
Problem solved for me, I had the same problem on several other smart switches with several channels. Now they work properly.
https://www.elektroda.pl/rtvforum/topic4026798.html

xjikka, 2024-02-02:
> Thank you very much! Problem solved for me, I had the same problem on several other smart switches with several channels. Now they work properly. https://www.elektroda.pl/rtvforum/topic4026798.html

I'm glad it helped. Enjoy.

openshwprojects, 2024-02-04:
Thank you, I've decided to apply your patch:
https://github.com/openshwprojects/OpenBK7231T_App/commit/a634e2b8b30f12eccd94e421d2c4fc8af0199176

closed 2024-02-04 by openshwprojects

xjikka, 2024-02-04:
> Thank you, I've decided to apply your patch: [a634e2b](https://github.com/openshwprojects/OpenBK7231T_App/commit/a634e2b8b30f12eccd94e421d2c4fc8af0199176)

Thanks

rt400, 2024-08-18:
The problem is still in the latest firmware... maybe the patch remove ?
cause if i flash the firmware above the problem are gone

rt400, 2024-08-18:
@xjikka Can you share which change you made to make it work ?

xjikka, 2024-08-18:
> @xjikka Can you share which change you made to make it work ?

https://github.com/openshwprojects/OpenBK7231T_App/issues/1000#issue-2045233165
https://github.com/xjikka/OpenBK7231T_App

rt400, 2024-08-18:
So Strange...
I see this patch in the last firmware but its not work...
if i flash your rbl file that you build its work fine

xjikka, 2024-08-18:
> The problem is still in the latest firmware... maybe the patch remove ? cause if i flash the firmware above the problem are gone

My patch is done about 3 months, but it is still waiting for merge, I don't know why it take so long, maybe holiday stuff.
I have no rights to merge it to master. For merge to master please ask @divadiow  @openshwprojects 
https://github.com/openshwprojects/OpenBK7231T_App/pull/1274#issuecomment-2278991262

xjikka, 2024-08-18:
@rt400 oh, sorry, You talking about the TglChanOnTgl, not about SSID2. If you want, you can check my last build from 5/2024  here: 
[OpenBK7231N_dev_20240527_195910.zip](https://github.com/user-attachments/files/16651193/OpenBK7231N_dev_20240527_195910.zip)

divadiow, 2024-08-18:
> or merge to master please ask @divadiow 

I'm afraid I have no special access

xjikka, 2024-08-18:
@divadiow sorry, I didn't know. The only one is then @openshwprojects  - One man to rule them all, One man to find them, One man to bring them all, and in the OpenBK7231T_App.master bind them  ;-)

rt400, 2024-08-20:
> @rt400 oh, sorry, You talking about the TglChanOnTgl, not about SSID2. If you want, you can check my last build from 5/2024 here: [OpenBK7231N_dev_20240527_195910.zip](https://github.com/user-attachments/files/16651193/OpenBK7231N_dev_20240527_195910.zip)

with your build the problem with TglChanOnTgl is exists like the official firmware (the problem that on boot the relay become ON automatically)

rt400, 2024-08-20:
> @gutek18a Sure, here it is /rbl for web App OTA update/ [OpenBK7231N_dev_20231217_145714-rbl-webapp-ota.zip](https://github.com/openshwprojects/OpenBK7231T_App/files/14130765/OpenBK7231N_dev_20231217_145714-rbl-webapp-ota.zip) Note this: Compiled binary is based on source from 17.12.2023

this build work good and relay stay off on boot
so what the different ?

xjikka, 2024-08-20:
@rt400 Since then there have been a lot of other code changes that have affected this simple delayus(2000) solution. I don't have that Beken with ADC anymore, so I can't confirm that. You can try extending the delay to e.g. 10000 us, maybe it will help.

xjikka, 2024-08-20:
@rt400 the only way to find out which change affected it is to try release https://github.com/openshwprojects/OpenBK7231T_App/releases/tag/1.17.448 - that's where it was first added in master, and then selectively find that release, when it stopped. Then compare these 2 releases and find out where the difference is that affects it.

xjikka, 2024-08-20:
@rt400 I also thought that it could easily be affected by some change that occurred between 12/2023 (when I changed it) and 2/2024 (when it was added to master). I used my release on that cube with ADC (before it died), so it cannot be ruled out that the first master release also worked with that patch.

rt400, 2024-08-20:
@xjikka It's really weird..
I flashed the official version 1.17.448 which has your changes but the problem exists.
The problem only goes away if I flash your BUILD "OpenBK7231N_dev_20231217_145714-rbl-webapp-ota.zip"

For now I think I'll stick with this version of yours OpenBK7231N_dev_20231217_145714-rbl-webapp-ota.zip

xjikka, 2024-08-20:
@rt400 so the change in master never worked. Maybe it's a different SDK I used to compile, or changes in the code between 12/2023 and 2/2024. When I have some time I'll try to look at the differences to see if I can find anything.

xjikka, 2024-08-20:
@rt400 can you try release 451 https://github.com/openshwprojects/OpenBK7231T_App/releases/tag/1.17.45 ?
In the releases 443 - 450 was patch in source missplaced  - see https://github.com/openshwprojects/OpenBK7231T_App/commit/9262ab06853a03e21a0fc531486f408bcc2a69af

xjikka, 2024-08-20:
@rt400 and I compiled another test flash :

OpenBK7231N_App_20231217-recoompiled-20240820.rbl - original patched version from 20231217 newly compiled by me

OpenBK7231N_App_663-recompiled.rbl - current master 663 newly compiled by me

OpenBK7231N_App_663-recompiled-delay-20000.rbl - current master 663 + extended usleep to 20000 newly compiled by me

[OPENBKT.ZIP](https://github.com/user-attachments/files/16677883/OPENBKT.ZIP)

rt400, 2024-08-21:
@xjikka Here is me results : 
OpenBK7231N_App_20231217-recoompiled-20240820.rbl - Work OK with no turn on on boot

OpenBK7231N_App_663-recompiled.rbl - Not work, on boot the relay turn on

OpenBK7231N_App_663-recompiled-delay-20000.rbl - Work OK with no turn on on boot

xjikka, 2024-08-21:
@openshwprojects regarding to @rt400 test, I suggest to change delay from 2000 to 20000 in the new_pins.c
[new_pins.zip](https://github.com/user-attachments/files/16688545/new_pins.zip)
I don't have the ADC pin toggle switch anymore, so I can't verify that, but it is from rt400 confirmed.

xjikka, 2024-08-21:
> @xjikka Here is me results : OpenBK7231N_App_20231217-recoompiled-20240820.rbl - Work OK with no turn on on boot
> 
> OpenBK7231N_App_663-recompiled.rbl - Not work, on boot the relay turn on
> 
> OpenBK7231N_App_663-recompiled-delay-20000.rbl - Work OK with no turn on on boot

Thank You for testing

xjikka, 2024-09-24:
@rt400 It is [done](https://github.com/openshwprojects/OpenBK7231T_App/issues/1349#issuecomment-2371585019)

rt400, 2024-09-24:
> @rt400 It is [done](https://github.com/openshwprojects/OpenBK7231T_App/issues/1349#issuecomment-2371585019)

Thanks.
I will upgrade and reply if its OK

rt400, 2024-09-24:
@xjikka WOW now its work...
Thanks you for your help