openshwprojects/OpenBK7231T_App #1003: [Feature request] Set channel based on http response

callumgare, 2023-12-20:
My understanding is that the two HTTP request commands, sendGet and sendPost, currently ignore the response returned by the server. I want to be able to set a channel based on the response body.

There's a few ways I could imagine this working. Maybe when sending the request you would supply the channel that you want the response to be saved into. Then if the body of the request is just an integer this is saved into the channel (otherwise if the request is anything else it is ignored). 

Or alternatively maybe there could be a way to say "if the response matches this string then set the channel to this or if it matches this other string then set it to this other value".

The reason I want to read the response is because I want to make a light which sits at my friends house and periodicity checks a server via the internet to determine which colour it should display. The light needs to fetch the colour from the server rather than the server sending a request to the bulb to change since the bulb won't be accessible from the internet.

Thanks for reading and thanks for making such a great project! :)

openshwprojects, 2023-12-20:
Hey, maybe something like "get response and execute is as command"?

openshwprojects, 2023-12-20:
Maybe we can add it as a flag for sendGet? Make it disabled by default...

openshwprojects, 2023-12-23:
For now, I've added this:
https://github.com/openshwprojects/OpenBK7231T_App/commit/73e9b8f471b6804d804b20fce5e8d92530b678ab

callumgare, 2023-12-23:
Great! Thanks very much :)

openshwprojects, 2023-12-24:
You're welcome! if something is not working in a stable manner, please open next issue

closed 2023-12-24 by openshwprojects