openshwprojects/OpenBK7231T_App #1006: Building Instructions for BK7231N

sebska, 2023-12-22:
I there, 
i am new to using CYGWIN and i was trying to follow your Giude to build a tool for flashing a CB2S. (on a Windows 11). Its is a bit hard to follow when you have only small experience with unix or flashing devices.

I was following this page: "https://github.com/haemka/OpenBK7231-Antela-F1s203-EU"

**Describe the bug**
I was fllowing the page "https://github.com/openshwprojects/OpenBK7231T_App/wiki/Building".
I found some issues. You shold mention, which Packets in cygwin need to be installed (i found git and make - are there more?).
You should open Cygwin as a administrator.
I hopefully managed to download the SDK. You need only to download ist, not to compile?
In the Code-section of "2) Install the app repository" there  ist a fault: the text "git clone https://github.com/openshwprojects/OpenBK7231T_App apps\OpenBK7231T_App"  needs to be "apps/OpenBK7231T_App" (not a "\" but a "/").

Now i am stuck at the point "3) Build the project"

Neither of both options works. "b.sh" was first not executable, but within it uses the phrase "OpenBK7231N_App" but there is only a "OpenBK7231T_App". Perhaps it usese the directory name.
So i used my on phrase:  ./build_app.sh apps/OpenBK7231T_App OpenBK7231T_App 1.0.0

His happened:

APP_PATH=apps/OpenBK7231T_App
APP_NAME=OpenBK7231T_App
APP_VERSION=1.0.0
USER_CMD=
Calling original build_app.sh script
cd /cygdrive/c/OpenBK/OpenBK7231N/apps/OpenBK7231T_App/../../platforms/bk7231n/toolchain/../bk7231n_os && sh build.sh OpenBK7231T_App 1.0.0 bk7231n
APP_BIN_NAME=OpenBK7231T_App
APP_VERSION=1.0.0
TARGET_PLATFORM=bk7231n
USER_CMD=
Start Compile
system:CYGWIN_NT-10.0-22000
not is ci build
make[1]: Entering directory '/cygdrive/c/OpenBK/OpenBK7231N/platforms/bk7231n/bk7231n_os'
fatal: No annotated tags can describe '7a65b01f637203c1c6c1f7a7ec630aa16ea5df4b'.
However, there were unannotated tags: try --tags.
make[2]: Entering directory '/cygdrive/c/OpenBK/OpenBK7231N/platforms/bk7231n/bk7231n_os'
echo prerequirement0
prerequirement0
===========================================================
Build OpenBK7231T_App
===========================================================
echo prerequirement1
prerequirement1
mkdir -p Debug/obj
mkdir -p Debug/bin
echo prerequirement2
build beken378/app/app_bk.o
prerequirement2
/bin/sh: line 1: ../toolchain/windows/gcc-arm-none-eabi-4_9-2015q1/bin/arm-none-eabi-gcc: Permission denied
make[2]: *** [application.mk:1178: beken378/app/app_bk.o] Error 126
make[2]: *** Waiting for unfinished jobs....
make[2]: Leaving directory '/cygdrive/c/OpenBK/OpenBK7231N/platforms/bk7231n/bk7231n_os'
make[1]: *** [Makefile:7: mp] Error 2
make[1]: Leaving directory '/cygdrive/c/OpenBK/OpenBK7231N/platforms/bk7231n/bk7231n_os'
make: *** [Makefile:6: all] Error 2

What can I do to flsch my device?
Best wishes Sebastian

openshwprojects, 2023-12-23:
I think you should rename directory to N for N platform. 

Are you compiling as root? Maybe you need to chmod full permissions recursively for SDK? If it's just a virtual cygwin environment, it should not hurt...

Btw why are you compiling yourself instead of getting automatic builds?

sebska, 2023-12-23:
Thanks for the reply.

Soit seems that i havent fond something about automatic builds or i have just overseen it...
Where can i find something about it and a step by step decription?

Sebastian

openshwprojects, 2023-12-23:
Do you want to just download build or want to build yourself?

We have releases tab on the right...

sebska, 2023-12-23:
just download and flash my device...

openshwprojects, 2023-12-23:
So, on the right, you have "Releaes" tab.... do you? 
And here is our flasher: https://github.com/openshwprojects/BK7231GUIFlashTool

sebska, 2023-12-30:
Worked, i did not find the flasher....
Now i have to calibrate the plug, but that should be no problem...

closed 2023-12-30 by sebska