openshwprojects/OpenBK7231T_App #1007: ISNATCH HeySwitch

davidebeatrici, 2023-12-22:
[Scheda tecnica 68.6810.65_1.pdf](https://github.com/openshwprojects/OpenBK7231T_App/files/13755233/Scheda.tecnica.68.6810.65_1.pdf)
[HEY SWITCH 68.6810.65 Manuale ITA-ENG.pdf](https://github.com/openshwprojects/OpenBK7231T_App/files/13755232/HEY.SWITCH.68.6810.65.Manuale.ITA-ENG.pdf)
[cod. 68.6810.65 - Dichiarazione di Conformità_1.pdf](https://github.com/openshwprojects/OpenBK7231T_App/files/13755231/cod.68.6810.65.-.Dichiarazione.di.Conformita_1.pdf)

<details>
<summary>Pictures</summary>

[image]
[image]
[image]
[image]
[image]
[image]
[image]
</details>

The MCU appears to be a `BK7231Q`: https://docs.libretiny.eu/boards/wa2

I'll attempt to perform a flash and report the result here.

davidebeatrici, 2023-12-22:
Oh, I just realized only the `BK7231N` and `BK7231T` are mentioned as supported by this project... can anyone confirm?

openshwprojects, 2023-12-23:
there is no known project supporting WA2 as fas a I know, but you can post your results there:
https://www.elektroda.com/rtvforum/viewtopic.php?p=20859869#20859869

closed 2023-12-23 by openshwprojects

davidebeatrici, 2025-08-06:
I ended up replacing the module with an `ESP8685-WROOM-03`:

[image]

And also replaced the button with an external one:

[image]