openshwprojects/OpenBK7231T_App #1009: Energy counter loses precision after 8.192 kwh

giedriuslt, 2023-12-24:
Energy for bl0937 is counted ant stored in Wh using float data type (4 bytes). However, it has quite little precision, and after 8192 wh it stops increasing with smallest power possible. 

After 8192Wh float has minimum delta of 0.000976562Wh. Minimum power bl0937 typically detects with sockets is around 1W. Over one second update interval its 1/3600 = 0.00027Wh , which is less than half of minimum delta, and so energy counter can't increase.

To fix this calculation could be changed to use double type instead of float. Or if double is not supported then proper way to sum such small numbers would be to use https://en.wikipedia.org/wiki/Kahan_summation_algorithm

openshwprojects, 2023-12-24:
Thank you for bringing this to my attention! I think double should work, I will change it to double, it's just... this will require a change in config structure, as this was not predicted when our contributor Valek added energy counting. Hmm

giedriuslt, 2023-12-24:
It can be changed to double only in memory, storage can still be float to preserve backwards compatibility. It would only cause problems if you disconnect and reconnect socket every few seconds, but sockets usually stays connected for a long time, so it would be fine.

openshwprojects, 2023-12-24:
@giedriuslt  can you check this version?
https://github.com/openshwprojects/OpenBK7231T_App/commit/d2ce54f0bfa069d8b608db95a7decf7264e24fc7
Would that be enough to increase precision for now?

giedriuslt, 2023-12-24:
Thank you, installed in one of the socket, will see overnight if it works correctly.

giedriuslt, 2023-12-25:
Can conform that now it counts correctly, thank you. Issue can be closed.

openshwprojects, 2024-04-05:
Ok, I see no issue reports related to that so I think I can close it.

closed 2024-04-05 by openshwprojects