openshwprojects/OpenBK7231T_App #1011: No WiFi connection after changing the defaults

onki69, 2023-12-26:
I flashed my BL 602 LED controller severasl times and I always got to the default WiFi connection.
When I entered my WiFi settings (checked many times and copied through copy/paste) the device reboots and no WiFi is present anymore.
Is there a kind of emergency procedure to get back to the default WiFi or what are the UART parameters to check the oputput at the UART console?

openshwprojects, 2023-12-26:
Is your WiFi 2.4GHz? 

You can do 5 quick power off/on cycles, see FAQ

onki69, 2023-12-26:
Yes sure it is 2G4. 
I have flashed many Tasmota devices and did the same way (default WiFi) with them.
Any Info on the serial terminal to see any error messages?

onki69, 2023-12-26:
[20231226182752.log](https://github.com/openshwprojects/OpenBK7231T_App/files/13772919/20231226182752.log)
This is a log file through the terminal (2MBaud). My PSK is replaced with "deleted".

openshwprojects, 2023-12-27:
Can you try with another router? Is your router a mesh network?

onki69, 2023-12-27:
My Router (Fritzbox) is in a meshed environment but I don't have another router available.