openshwprojects/OpenBK7231T_App #1012: MQTT host, build, mac sent at boot, but not sent later, and not retained

grunthos503, 2023-12-28:
OBK transmits several device values to MQTT when it boots up, including "host", "build", and "mac"

If you have Flag 2 turned on, OBK will transmit other state items periodically, such as rssi and ssid, but not host, build, and mac.   If an MQTT listener connects after the OBK device has booted, it will receive other state values, but not these.  There is no way for a later listener to find out these values

**Firmware:**
- Version: 1.17.375 (Build on Dec 24 2023 13:30:13)
- Device:  Minoston MT10W timer switch
- Chip/model: BK7231N

**To Reproduce**
Steps to reproduce the behavior:
1. Enable flag 2
2. Start an MQTT listener for topic "devicename/build"
3. Reboot device
4. Receive MQTT message with build info string
5. End MQTT listener, then subscribe again
6. Periodically device will transmit other state values on MQTT, but will not send "devicename/build" topic

In-wall mains-powered OBK devices may almost never get rebooted, so most MQTT listeners will never receive the device info.

**Additional context**
This assumes flag 7 (retain all) is off

Possible fixes:  
1. Transmit during state update, like ssid and rssi.
2. Transmit with retain flag for these values

openshwprojects, 2023-12-28:
Do you have this flag on:

10 | [MQTT] Broadcast self state on MQTT connect

grunthos503, 2023-12-28:
It behaves the same, whether I have flag 10 on or off.   Whether flag 10 is on or off, I only see it transmit host, mac, and build when it first boots up.  If I use the Restart button in the web UI, it will reboot, and then it does transmit them.  After that, they are no longer sent.  If I connect to broker with MQTT Explorer, or restart HA, I don't receive any new updates of these topics.

Flag 10 says "on MQTT connect".  That would be when the device connects to the broker, right?  (As I understand MQTT, the device doesn't know when other listeners connect to the broker.)  Looking at client connection count on my broker (Mosquitto), it appears that my device connects to the broker when it boots, and then stays connected.  If it doesn't reconnect, then it won't send the flag 10 items, I expect.

icovelev, 2024-04-01:
Similar issue, build version became unknown after Home Assistant restart, so the data is not transmitted with all the other attributes every 60 seconds as per flag 2.
[image]