openshwprojects/OpenBK7231T_App #1013: TuyaMCU driver - Long data packets possibly cause overflow

Angel0ffDeath, 2023-12-28:
Hi I described almost everything here: https://www.elektroda.com/rtvforum/viewtopic.php?p=20880051#20880051
but in order not to read the whole post this is the extraction:

 received: 55 AA 03 07 00 10 12 00 00 0C 01 01 00 3F 03 01 00 FA 04 01 00 AA 25 
Info:TuyaMCU:TuyaMCU_ProcessIncoming[ver=3]: processing command 7 (State) with 23 bytes
Info:TuyaMCU:TuyaMCU_ParseStateMessage: processing dpId 18, dataType 0-DP_TYPE_RAW and 12 data bytes
Info:TuyaMCU:TUYAMCU received: 55 AA 03 07 00 08 0E 02 00 04 00 00 00 64 89 

55 AA 03 07 00 10 12 00 00 0C 01 01 00 3F 03 01 00 FA 04 01 00 AA 25

Data -> 01 01 00 3F 03 01 00 FA 04 01 00 AA - Data is correct - 63A(3F), OV=250V, UV=170V break the line for all 3 events

Im pretty sure if tuyaMCU driver tries to make integer from these 12 bytes (like for dpID 17) there will be overflow. Even longint will overflow, so this is a problem of tuyaMCU driver.

As a result this field is always 0.

The device is TOMPD-63WIFI with WB3S chip; Flashed firmware OpenBK7231T_UA_1.17.376.bin
Definition of the field we speak is
// Alarm set 2 - Dpid 18 "alarm_set_2" - channel 18
linkTuyaMCUOutputToChannel 18 raw 18
setChannelType 18 textfield
setChannelLabel 18 "Alarm set 2"

About reproduction of the error - only if you have device sending so long data packets.
Everything else seems to work more or less ok.

openshwprojects, 2023-12-28:
I'll reply on forum, I don't check git as often as forum.

Angel0ffDeath, 2024-01-19:
It looks like there is no development on this issue

openshwprojects, 2024-04-05:
Some time has passed and I am not sure, is it solved now, or does the problem still persist?