openshwprojects/OpenBK7231T_App #1014: Flashing WB2-12F dev board with OpenBeken

MacSass, 2023-12-31:
Hello,
I ordered two WB2-12F development boards, hoping I would be able to flash them with OpenBeken.
This is the board: https://de.aliexpress.com/item/1005004911487557.html 

As per my understanding those boards are running a BL602 chip, if I´m not mistaken. Are those supported by OpenBeken (I think so) and do you know if those boards can be flashed through the USB port located on the development board.

Thanks for helping an OpenBeken newbie ...

openshwprojects, 2023-12-31:
Hello, have you tried BL602 guide?
https://www.elektroda.com/rtvforum/topic3889041.html
Or a video tutorial? Search for BL602 here:
https://www.youtube.com/@elektrodacom

MacSass, 2023-12-31:
Thanks - I saw that post, but I was hoping it was possible to flash my board through the on-board USB/Serial converter ... but that does not seem to work?
The board mentioned in the post linked above or in the videos are very different and does not have an on-board USB converter.
Looks like my board is fairly new and has not gained a lot of visibility yet ...
I will try connecting my own serial converter to the respective PINs on my board directly to see if I can flash it that way ...

openshwprojects, 2023-12-31:
But are you using correct flash tool? DL Dev Cube? I saw your post in BK7231 flasher git, but BK7231 flasher is for BK7231 platform, not for BL602 hmm

MacSass, 2023-12-31:
Ah, ok - thank you, I might have missed that one ... I saw in the one I used mentioned "WB2" ... guess I´m still learning what is what in OpenBeken.
I take a look at the DL Dev Cube. 
Thanks again for pointing that out to me!

MacSass, 2023-12-31:
With DevCube I was able to flash the board through the on-board USB, after hitting the "burn" button while powering on - worked like a charm.
Thanks for pointing me in the right direction! Now I just need to figure out how to configure the on-board RGB LED in OpenBeken - but this is something for the new year.
Thanks again for the help and have a good start!

closed 2023-12-31 by MacSass