openshwprojects/OpenBK7231T_App #1020: TR6260 support

salvomaltese, 2024-01-03:
I have purchase a smart switch from Aliexpress, but unlike what I expected the chip is not a BK7231N/T or similar but a TR6260.
By researching the web I found the following information:

https://www.elektroda.pl/rtvforum/topic3960571.html
https://github.com/Akeeta/Akeeta-SDK-TR6260/tree/master
https://github.com/Akeeta/Akeeta-SDK-TR6260/blob/master/%E9%9B%85%E8%A7%82tr6260-SDK%E4%BD%BF%E7%94%A8%E8%AF%B4%E6%98%8E.pdf
http://www.skylab.com.cn/newsview-697.html

It seems to be based on esp8266 or similar.
is it possible to create a build with openbeken for this chip?

Thanks so much in advance :)

openshwprojects, 2024-01-03:
Hello, is there an UART flash tool for that?

salvomaltese, 2024-01-03:
I couldn't find anything about it

salvomaltese, 2024-01-15:
Finding on web I have found at https://electric.garden/shenzhen-leadcoo-2axje/wifi-module-la-we3l this TEST report https://electric.garden/shenzhen-leadcoo-2axje/test-report/LA-WE3L-Test-Report-Shenzhen-Leadcoo-2axje-la-we3l-ex-1-9.pdf
Where they used this tool "UTP"
[image]

openshwprojects, 2024-01-15:
but do we have a download link?

salvomaltese, 2024-01-16:
I have requested the AURT flash tool to Skylab
http://www.skylab.com.cn/productview-150-TR6260_uartwifi_lcs6260.html
Will follow up with updates when I hear back from them

jekader, 2024-01-17:
This output looks like a standard Tuya CLI since TR6260 is listed as supported by their platform.
https://developer.tuya.com/en/docs/iot-device-dev/hardware?id=Kaiuyfgqxi1cl

Looking at the datasheet for this chip - looks like an Andes N25F RISC-V core or something similar from the same family:
https://fcc.report/FCC-ID/2AUUL-TR6260/4520594.pdf
CPU specs look quite similar, especially the PowerBrake part:
https://www.andestech.com/en/products-solutions/andescore-processors/riscv-n25f/

There is no openly available SDK, I only found sources for the 6600U wifi dongle also from Eswin - that one looks like an MCU pretending to be a USB device, so maybe its firmware upload code can be useful to find the commands needed for flashing the other chip:
https://github.com/eswincomputing/eswin_6600u/blob/master/usb/fw.c

jekader, 2024-01-22:
After some more digging, found a couple firmware images for TR6260 and they seem to point to this SDK from andestech:

https://github.com/andestech/Andes-Development-Kit/releases/tag/ast-v5_2_1-release-linux

Looking at the compatibility table in [AndeSight_STD_v5.3_IDE_User_Manual_UM271_V1.0.pdf](https://fpga.andestech.com/site/2/document/download/AndeSight_STD_v5.3_IDE_User_Manual_UM271_V1.0.pdf), `nds32le-elf-[newlib|mculib]-v5` is the SDK to use with this MCU.

Source code for the toolchain to be available here:
https://github.com/andestech/nds-gnu-toolchain

Should be possible to port OpenBeken to TR6260, someone with enough skills would just need to have the hardware on hand. I see HLK-M20 for sale so if there's anyone willing I can sponsor some boards just for the fun of it. Not sure if it's worth the effort though as this MCU is not very ubiquitous and superceeded by more modern hardware.

openshwprojects, 2024-01-23:
What about ther flash tool? I would love to try porting. I am currently finishing LN882H support, and TR6260 could be next.

jekader, 2024-02-08:
Got HLK-M20 boards in the mail this week which sparked my interest again. So far could not get them to output anything. Did take a look through AndeSight docs and I don't think they will be of much help. They only cover the core, not the peripherals (WiFi, GPIO, etc) of the specific implementation.

Found some hints of what the SDK is called in the [Akeeta-SDK-TR6260](https://github.com/Akeeta/Akeeta-SDK-TR6260/blob/master/%E9%9B%85%E8%A7%82tr6260-SDK%E4%BD%BF%E7%94%A8%E8%AF%B4%E6%98%8E.pdf) repo:

SDK version: TR6260V1.1.1B01P02
Manufacturer's partition table:
[image]

Some of the HLK-M20 images I've obtained contain a different parititon structure, including OTA and MQTT support. Since there's no known flasher tool at this point, I don't think an alternative firmware is possible until the full SDK is published.

Some strings which could be used to search for the SDK:

platform/otaHal/otaDiffHal.c
wifi/hal/tr6260/hal_phy_tr6260.c
wifi/mac/lmac/hal_lmac_common.c

If there is any interest in reverse-engineering the firmware images, extracting the bootloader, etc - let me know, I can share some binaries.

On a side note, some other modules using this MCU: 
* LCS6260 UART to WiFi module similar to HLK-M20
* XY-WE2S, LA-WE2S Tuya IoT modules (note, [TYWE2S](https://developer.tuya.com/en/docs/iot/wifie2smodule?id=K9605u79tgxug) has a similar name and form facotr but is based on ESP8285)

divadiow, 2024-11-05:
> If there is any interest in reverse-engineering the firmware images, extracting the bootloader, etc - let me know, I can share some binaries.

I wouldn't mind a little look please :)

I've posted some HLK-M20 datasheets here https://www.elektroda.com/rtvforum/viewtopic.php?p=21290001#21290001

divadiow, 2024-11-19:
@jekader

divadiow, 2024-11-19:
SDK, burner and toolchain also posted

divadiow, 2024-11-19:
cool. im about to post something else exciting too :)

jekader, 2024-12-13:
This issue can be closed now since #1446 is merged

divadiow, 2024-12-14:
flashtool repo also https://github.com/openshwprojects/FlashTools/tree/main/TransaSemi-ESWIN