openshwprojects/OpenBK7231T_App #1031: Add float support for Temperature

enesbcs, 2024-01-14:
**Describe the bug**
SetChannelFloat sets value correctly, but "temperature" channel remove decimals.

**Firmware:**
- Version 1.17.407
- Device Neo NAS-AB02W6 v4
- Chip/model: BK7231N
- Device config TuyaMCU

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Apps-Log-Command:
```
setChannelType 2 Temperature
SetChannelFloat 2 23.5
```
5. See error
On Main page it displays:

> Temperature 23.0C (2)
> Channel 1 = 0, Channel 2 = 23.50, 

(And it forwards 23 through MQTT also.)

enesbcs, 2024-01-14:
Although i could use "Temperature_Div10" channel type but it sends 235 through MQTT which is not really liked by Domoticz.. as it has basic discovery support, except no template support.

The device full autoexec.bat is the following:

```
//Drivers
startDriver TuyaMCU
//startDriver NTP
//ntp_timeZoneOfs 1:00
//
//Channels
setChannelType 1 toggle
// report float as Domoticz has no template support!!!
setChannelType 2 Temperature
setChannelType 3 Humidity
//ch4 = chime sound
setChannelType 4 TextField
//ch5 = volume 0->high 1->middle 2->low
setChannelType 5 TextField
//ch6 = alarm duration in sec
setChannelType 6 TextField
//dummy channel
setChannelType 7 Temperature_Div10
// Hide unneeded raw data
SetChannelPrivate 7
SetChannelVisible 7 0
//
//Links & Vars
linkTuyaMCUOutputToChannel 104 bool 1
linkTuyaMCUOutputToChannel 106 val 3
linkTuyaMCUOutputToChannel 102 enum 4
linkTuyaMCUOutputToChannel 116 enum 5
linkTuyaMCUOutputToChannel 103 val 6
linkTuyaMCUOutputToChannel 105 val 7
//Stop blinking
tuyaMcu_defWiFiState 3
//Ask for status at startup
tuyaMcu_sendQueryState

//
//Events
//Copy temp to channel2 when data arrives
addEventHandler TuyaMCUParsed 0x07 backlog SetChannelFloat 2 $CH7/10;publishChannel 2;
//Report temperature data at every 2 minutes
addRepeatingEvent 120 -1 backlog tuyaMcu_sendQueryState;

```

openshwprojects, 2024-01-14:
I think you can enable flag to apply divider before MQTT publish and it will send 23.5

enesbcs, 2024-01-14:
> I think you can enable flag to apply divider before MQTT publish and it will send 23.5

It sounds interesting, which Flag does this?

closed 2024-01-14 by enesbcs