openshwprojects/OpenBK7231T_App #1033: New LSC Smart Mood light 3204432

snakesrules94, 2024-01-14:
Hi all and happy new year to the community ! 
I bought today some new smart light from action which afaik is not yet seen on the database :

[image]
[image]

opening is quite easy by gently cutting the glue around the edge of the refractor and then it comes :

[image]
[image]

This time , the BK7231N is soldered directly :

[image]
[image]
[image]

The light is recognised in tuya app with fw 2.0.0.

I've tried to flash with TCC by using a 2.0.0 7231N strip profile but without luck ( also no way with the others 2.0.0 7231N profiles ).
It seems that  the line of 9 solder points on the back of the pcb could be connected for somes to the MCU and could be used as direct access for flashing but i have no time right now to check , i will be off for 15 days

There is also a microphone on the board as this lamp can react to music .

[image]
[image]

Regards.

snakesrules94, 2024-01-14:
I've tried quickly to dump the fw with the lightleak app using the differents profiles but no way. Will go deeper with this stuff when i'll be back.

openshwprojects, 2024-01-14:
Thank you for sharing, but usually, we post our teardowns here:
https://www.elektroda.com/rtvforum/forum51.html
If you want to continue on Github, then ok, just let me know, can I repost your teardown (once it's complete - with template) to our forum?

snakesrules94, 2024-01-15:
Hello,
Sure ! You can repost on the forum. I’ve just wrote here because i have no account yet on electroda. Will continue on that after my trip. 
rgds.

openshwprojects, 2024-01-19:
Let me know once you get this device flashed and working. Good luck

snakesrules94, 2024-01-28:
Hello, here are the latests news .
Lamp is now flashed and running the latest Openbeken fw.
I've made a push request to TCC for integrating the flashing profile.

template :
`{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Lsc Smart Mood Light",
  "model": "3204432",
  "chip": "BK7231N",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "6": "PWM;1",
    "7": "PWM;2",
    "14": "Btn_Tgl_All;0",
    "24": "PWM;3",
    "26": "PWM;4"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}`

snakesrules94, 2024-01-29:
Final template :
`{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Lsc Smart Mood Light",
  "model": "3204432",
  "chip": "BK7231N",
  "board": "TODO",
  "flags": "1024",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "6": "PWM;1",
    "7": "PWM;2",
    "14": "Btn;5",
    "24": "PWM;3",
    "26": "PWM;4"
  },
  "command": "",
  "image": "https://obrazki.elektroda.pl/YOUR_IMAGE.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic_YOUR_TOPIC.html"
}`

Now back power button is working fine.

openshwprojects, 2024-04-05:
It's now present here:
https://www.elektroda.com/rtvforum/topic4033331.html
Thank you, closing as solved

closed 2024-04-05 by openshwprojects