openshwprojects/OpenBK7231T_App #1034: Web app doesn't work with domain name (needs IP address_

joe-sydney, 2024-01-14:
**Describe the bug**
A clear and concise description of what the bug is.
Hi there, I am not sure this is a bug or not, but I am sure there was a change in behaviour along the way.

Before 1.17.366, I was able to access my lights using their domain names (eg light1.mydomain.net). Then I was able to launch the web application from the main page, and all is ok with the web app.

I have upgraded to latest 408, and the same situation remains.

Now, I can still access the lights with their domain names, but when I click on Launch Web Application, I get the following error: Error: TypeError: NetworkError when attempting to fetch resource.

By trial and error, I was able to figure out finally that using the native IP address (rather than the domain name) was fine.

**Firmware:**
- Version [e.g. 1.14.129] approx 1.17.366 and after 
- Device? 
- Chip/model: BK7231T?
- Device config?

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

openshwprojects, 2024-01-15:
Hello, can you narrow a bit more which revision broke this for you?

joe-sydney, 2024-01-15:
It is hard to say, as I jumped a few releases to 366. I could start from 366 and downgrade one release at a time till it works again, but I am afraid of backward compatibility issues of the configs from downgrading...

openshwprojects, 2024-01-16:
You can always just download LittleFS backup, template, and maybe config binary (just remember the version you are on).

I am asking about this because as far as I know there was no changes made for Web App during last months, or, dare I say, last year, that could change the hostname/IP behaviour.

So, it would be much easier for me to check what went wrong if I had know which revision broke that.