openshwprojects/OpenBK7231T_App #1036: Wifi RSSI not retain on deepsleep

erdeidominik99, 2024-01-17:
**Describe the bug**
On devices with deep sleep enabled, flag 7 (retain) is enabled, the wifi rssi is not retained, it shows not available after the chip goes to deep sleep , but the another values works good.

**Firmware:**
- Version: 1.17.409 (also not works with older)
- Chip/model: BK7231N

openshwprojects, 2024-01-19:
That's strange, I did a brief investigation today and it seems that RSSI publish goes through the same call as all MQTT publishes so OBK_FLAG_MQTT_ALWAYSSETRETAIN flag (7) should work. This requires futher investigation

erdeidominik99, 2024-01-19:
I looked at the history, the value is available for 3 mins after publish, then unavailable.

openshwprojects, 2024-01-20:
maybe you need to check the flag to disable avty-topic?

erdeidominik99, 2024-01-20:
Flag 35 is already checked.