openshwprojects/OpenBK7231T_App #1038: Can't set mac adress on BL602 device

hojnikb, 2024-01-17:
I'm using a smart plug based off BL602. When i flashed OpenBeken on it, it wiped the MAC adress (as expected). Now i have a few of these devices and they all have the same MAC adress by default. No problem.

But when i try to set a different MAC adress in the settings, it doesn't actually set. "Device MAC" indicates changed MAC, but when i check my router, i still see the same MAC adress as before. And obviously when i plugin more than one device at a time, i get conflicts.

Any ideas how to fix this? I'm currently using version 1.17.366.

openshwprojects, 2024-01-17:
Why do you think it's expected? 
I flashed 3 or 4 BL602 devices, two of them here:
https://www.youtube.com/watch?v=L6d42IMGhHw&t=2s&ab_channel=Elektrodacom
https://www.youtube.com/watch?v=bs0ylC6xRs0&t=59s&ab_channel=Elektrodacom
and MACs are clearly different. Did you do the flashing in somewhat different way than we do?

Regarding MAC change itself - this is unimplemented so far, as I was sadly unable to find a way to do it. I have infact tested some functions from SDK but I bricked my BL602 board in a meantime (by using some kind of efuse mac set) so I am not able to look into it any futher. Any help and investigation on BL602 MAC issue is welcome

hojnikb, 2024-01-18:
> Why do you think it's expected? I flashed 3 or 4 BL602 devices, two of them here: https://www.youtube.com/watch?v=L6d42IMGhHw&t=2s&ab_channel=Elektrodacom [...]

I've followed the video in your post, but i used a newer version of lab dev cube. Perheps there was some setting, that set MAC adress while flashing? I have to check.