openshwprojects/OpenBK7231T_App #1049: Is it possible to wake up deep sleeping Tuya CBU either by time or button press ?

deepcoder, 2024-01-26:
I am interested in creating a battery powered device that will wake up either via the DeepSleep command set to 5 minutes or if a button is pressed on the device via PinDeepSleep command. Is it possible to have both events active on a CBU BK7231N Tuya device? I would like to repurpose a Door/Window device that does NOT have a Tuya MCU for this project.

How would I script this in the autoexec.bat ?

Thx!

openshwprojects, 2024-01-27:
We have it in SDK, but it's not yet integrated to OBK:
https://github.com/openshwprojects/OpenBK7231N/pull/18
I need to find a moment to do that, or would you like to try?

deepcoder, 2024-01-27:
Thank you for the reply, info and your work.

If by 'would you like to try' you mean testing a .bin firmware file with this function, yes I would be happy and interested to try it on the project I am attempting. Please let me know how I can effectively test it for you.

logon84, 2024-02-27:
I would like to see this implemented in OBK too, I keep an eye here.