openshwprojects/OpenBK7231T_App #1068: Add Button click(s) into HA's MQTT Device trigger using in Automation

ptiFisher0, 2024-02-08:
*New feature*
Add Button clicks into Home Assistant's MQTT Device trigger so that they can be used in Automation.

It should work same like if you put "SetOption73 1" in Tasmota, it will send all button clicks to HA (MQTT).
Using HA's [MQTT Device trigger](https://www.home-assistant.io/integrations/device_trigger.mqtt)  which is a part of the HA's auto discovery, we can create triggers into the Automation UI. 
After using that we can select triggers like ("Second button" double clicked) or ("Second button" continuously pressed) inside Automation UI.

I tested this and it works great in my use. (tested with *3-Gang Touch Wall Switch* and *Mini Smart Switch*)
I put my sample code in [GitHub branch](https://github.com/ptiFisher0/OpenBK7231T_App/tree/btn-ha-trigger).
It will set this action now only into pins with role **IOR_Button_ScriptOnly**.

openshwprojects, 2024-02-09:
From a brief glance it looks very cool. I think you can open a Pull Request.

I am just wondering whether we need a flag to control MQTT button discovery, because, maybe not everyone wansts them discovered? But I can;'t think of any possible scenario where somebody may not want that button discovered, so let's go without a flag for now.

ptiFisher0, 2024-02-09:
Good, for me this is very nice adds. 
And I really, really like this project :)
PR #1072

ptiFisher0, 2024-02-14:
PR is updated now.
I fix merge issue and made some other fix and adds too.