openshwprojects/OpenBK7231T_App #1074: The internal temperature reading is very high on BK7231T

Wolfieeewolf, 2024-02-10:
**Describe the bug**
The internal temperature reading is very high on BK7231T. Have them read over 150°C. I assume it is meant to be read in °C not °F.  Even if it was °F it would still be reading over 60°C which isn't normal.

Seems to not be an issue with BK7231N most chips reading at about 30°C

**Firmware:**
- Version 1.17.457
- Device - So far all my BK7231T devices are reading very high
- Chip/model: BK7231T

**To Reproduce**
Steps to reproduce the behaviour:
1. OTA to version 1.17.457

**Screenshots**

[image]

divadiow, 2024-02-10:
indeed

[](https://www.elektroda.com/rtvforum/viewtopic.php?p=20954018#20954018)

vothar, 2024-02-10:
same behaviour for me on 4 switches BK7231T with openBK that i moved from 1.17.366 to 1.17.457 with OTA update. read more than 140 Celcius degres on the config page

ryan-ronnander, 2024-02-14:
I've seen the same readings on my end. I was initially worried because I first noticed the temps with a lightbulb. But my power strip and dimmer are also showing high temps.

vothar, 2024-02-14:
temperature reading still incorrect in version 1.17.460 for BK7231T firmware

stefan064, 2024-02-28:
Saw this as well. Also noticed T devices actually are running noticably hotter than similar N ones. Not to the extreme extent reported but still seems abnormal.
Two energy metering plugs with BL0937, same brand; left one BK7231T on WB2S, right one BK7231N on CB2S
[image]

jenshenk, 2024-03-11:
I wonder if the issue is more about an unnecessary 1 added before the real temperature instead of it being real fahrenheit degrees. Mine are "only" showing  130, which would still mean around 60 degrees C which is really not the case. 

@stefan064 did you try to add "PowerSave 1" to the autoexec.bat? It can make devices run at least a bit cooler

stefan064, 2024-03-12:
Hi @jenshenk, , I believe I did have 'powersave 1' enabled on both devices. The leading 1 (i.e. offset off 100?) does seem realistic actually. I see 153°C and 154°C on my hot-running T devices.
I dont know if this abnormal heat is relevant and if it applies to all T chips or just mine, The temp reading is certainly way off either way.
I had a look though the SDKs comparing T with N and saw what looked like sample skipping/averaging in which I'd guess there is probably some wrong in the arithmetic there amonst all the conditionals...

jenshenk, 2024-03-12:
Interesting find in the SDK, maybe this could lead to a solution? About your hot T plug: Hard to tell whether it might just be a quality control issue, after all these plugs are dirt cheap. On the other hand if I remember the socket design right it looks like you are in Australia where it tends to get warmer than over here in the Netherlands :)

stefan064, 2024-03-19:
I saw recent firmware releases have some T-specific temp adjustment, my 'hot' T devices now report around 60°C, much more realistic, thank you openshwprojects.
Out of interest, power draw on 5V supply for these particular T devices below. This has a double-pole output using 2x relays driven together.
- relays on: 243mA, 1.2W
- relays on with neutral-side relay removed: 181mA, 0.9W
- relays off: 107mA, 0.54W
- relays off, powersave 1: 100mA, 0.50W
openshwprojects has a video demonstrating a far greater effect of powersave than this. Perhaps something is dodgy with the T suffix if not just my particular devices.
I've decided to just let these run and see if they at least remain reliable.

StefanRu1, 2024-04-10:
Hi, can you tell more about " T-specific temp adjustment". 
I have 4 SmartPlugs.
3 of them return a correct value for 
Internal temperature: 42.7°C

But 1 of them is reporting it wrong. 
It is starting with a negative number and when the temp gets hotter it is rising more towards 0.
So the start value is not 0. So i have to shift the temp.
Reading is Internal temperature: -33.1°C
So i have to add 75,8.
Is that possible and how do i do this?

Thanks and best regards,
Stefan

closed 2024-05-13 by Wolfieeewolf