openshwprojects/OpenBK7231T_App #1078: Home Assistant incorrect light color when using flag 0 + 24  (RBGW bulb)

ryan-ronnander, 2024-02-12:
Hello,

After getting acquainted with all of the flags and options needed to get RGBW bulbs working with an emulated color temp (`flag 24`), `flag 0` (which appears to be a default flag) needs to be disabled, otherwise Home Assistant thinks the bulb is still the last chosen color after picking a color temperature based "color".

Video of what happens in Home Assistant:

https://github.com/openshwprojects/OpenBK7231T_App/assets/61520/deddfa95-10ff-49da-92ac-d19a4d4a8183

If this is expected with the combinations of flags, feel free to simply close out the issue. Maybe a quick note somewhere about "Do not enable flag 0 with flag 24" is needed, or something along those lines?

openshwprojects, 2024-02-14:
It certainly requires a futher investigation. RGBW bulbs are pretty rare in my area so I never really had one. It's possible that something is missing in a code, some kind of check for RGBW condition. The flag 0 should not break things, it should be handled on firmware side. I will try to find time to look into it soon, but any help is also welcome

ryan-ronnander, 2026-09-06:
Retested on 1.18.310 with the same bulb and flags, can no longer reproduce.

closed 2026-09-06 by ryan-ronnander