openshwprojects/OpenBK7231T_App #1079: Wall switch touch resposiveness

brunohoff, 2024-02-12:
**Describe the bug**
First of all, congrats for the project. This is more a question/feeling about the wall switch touch responsiveness. I have a couple of Tuya wall switches from different manufacturers and different chips. I have a feeling that after flashing the OpenBK software the time between pressing the button on the touch panel and the light being tuned on/off increases. The same time doesn’t appear to increase when turning on/off using web panel. Another thing is that it seems that the pressure to the touch sensor work must increase if compared with the original Tuya software. I flashed Tasmosta into some wall switches and have the same feeling about these parameters. 
So, my question is if there is some configuration to increase the responsiveness or even calibrate the touch sensor pressure.

Thank you in advance.

**Firmware:**
- Version all
- Device: Wall switches
- Chip/model: BK7231T and BK7231N

openshwprojects, 2024-02-12:
Have you tried:
[image]
src; https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/faq.md

brunohoff, 2024-02-12:
That's exactly what I was looking for. Thank you very much. I tried looking at issues and Google and was on the FAQ the whole time. Thanks.

closed 2024-02-12 by brunohoff