openshwprojects/OpenBK7231T_App #1080: Local Time gets crazy on BL602

Flobul, 2024-02-12:
**Describe the bug**
Local Time on web page gets crazy after seconds.
On boot, when NTP is synced, time is correct:
[image]

But seconds later:
[image]

Seconds and minutes increment correctly though time after time.

**Firmware:**
- Build on Feb 9 2024 16:36:16 version 1.17.457
- Device : Unamed Socket (LSPA9 clone) BL0937
- Chip/model: BL602
- Device config : 
```
{
  "vendor": "Tuya",
  "bDetailed": "0",
  "name": "Smart Socket 20A",
  "model": "BL602/BL0937",
  "chip": "BL602",
  "board": "TODO",
  "flags": "69731332",
  "keywords": [
    "TODO",
    "TODO",
    "TODO"
  ],
  "pins": {
    "2": "Rel;1",
    "3": "BL0937CF1;0",
    "14": "BL0937CF;0",
    "17": "Btn;1",
    "20": "LED_n;1",
    "21": "BL0937SEL;0"
  },
  "command": "backlog startDriver BL0937; startDriver NTP; startDriver SSDP; ntp_timeZoneOfs 1:00 ",
  "image": "https://obrazki.elektroda.pl/9568515100_1707749274.jpg",
  "wiki": "https://www.elektroda.com/rtvforum/topic4020634.html"
}
```

**To Reproduce**
Steps to reproduce the behavior:
1. Boot this BL602 device and wait a few seconds.

**Additional**
NTP log on boot:
```
Info:NTP:Seconds since Jan 1 1900 = 3916743508
Info:NTP:Unix time  : 1707758308
Info:NTP:Local Time : 2024/02/12 17:18:28
```

giedriuslt, 2024-03-07:
It seems SSDP driver is corrupting something somehow. If I don't start that it works. After starting ssdp it becomes garbage. Interesting to note that unix time remains correct.

giedriuslt, 2024-03-25:
Made a fix for this here #1145

Flobul, 2024-04-01:
Fixes for my devices. Thanks.