openshwprojects/OpenBK7231T_App #1081: Ledvance Smart+ WiFi Plug current measure problem

simontoyra, 2024-02-12:
Hello!
I have problems measuring the correct current with the software. I calibrate current with 1,8A load and when i turn off load current still
shows 1,024A. Same problem with 2 different ledvance units. Current sensor is HLW8012. The temperature of the unit also seems a bit on high side. Power measurement seems to work
[image]

openshwprojects, 2024-02-13:
Have you done tuya config extraction? Are you sure you got SEL pin set correctly?

simontoyra, 2024-02-13:
Tx for helping me, i think i did config extraction, look below. Im almost sure SEL pin is set accordingly

[image]

{
	"sel_pin_pin":"26",
	"rl1_lv":"1",
	"bt1_pin":"6",
	"net_trig":"1",
	"jv":"1.1.4",
	"netled1_lv":"0",
	"bt1_type":"0",
	"nety_led":"2",
	"vi_pin":"8",
	"resistor":"1",
	"over_cur":"21000",
	"bt1qp$Bgw_di_lv":"0",
	"reset_t":"5",
	"netled1_pin":"10",
	"chip_type":"1",
	"lose_vol":"90",
	"over_vol":"276",
	"module":"WB2S",
	"ele_pin":"7",
	"ch_cddpid1":"9",
	"ch1_stat":"0",
	"ch_num":"1",
	"rl1_pin":"24",
	"netn_led":"2",
	"vol_def":"0",
	"ch_dpid1":"1",
	"sel_pin_lv":"1",
	"crc":"107",
	"auth_key":"X",
	"ap_s{abi":"0",
	"id":"null",
	"swv":"7.0.0",
	"bv":"30.05",
	"pv":"2.2",
	"lpv":"3.3",
	"pk":"X",
	"firmk":"X",
	"cadv":"1.0.2",
	"cdv":"1.0.0",
	"dev_swv":"7.0.0",
	"s_id":"null",
	"dtp":"0",
	"sync":"0",
	"attr_num":"0",
	"mst_tp_0":"0",
	"mst_ver_0":"null",
	"mst_tp_1":"0",
	"mst{uuid":"X",
	"psk_key":"X",
	"ap_ssid":"LDV SMART+",
	"ap_passwd":"null",
	"country_code":"CN",
	"prod_test":"false }_ver_1",
	"mst_tp_2":"0",
	"mst_ver_2":"null",
	"mst_tp_3":"0",
	"mst_ver_3":"null",
	"dminfo_name":"null",
	"dminfo_code":"null }{abi",
	"mstEp$Bgw_ai{key":"null",
	"lckey":"null",
	"h_url":"null",
	"hs_url":"null",
	"hs_psk":"null",
	"mqs_url":"null",
	"mq_url":"null",
	"ai_sp":"null",
	"mq_psk":"null",
	"time_z":"null",
	"s_time_z":"null",
	"wx_app_id":"null",
	"wx_uuid":"null",
	"dy_tls_m":"0 }4$Btimer_arr{lastFetchTime",
	"cnt":"0}_ver_1",
	"ssid":"null",
	"passwd":"null",
	"md":"0",
	"random":"0",
	"wfb64":"1",
	"stat":"0",
	"token":"null",
	"region":"null",
	"reg_key":"null }{lastFetchTime",
	"ap_s{sel_pin_pin":"26",
	"bt1qp0$Bgw_di_lv":"0"
}

openshwprojects, 2024-02-13:
can you try setting BL0937_SEL_n role?

simontoyra, 2024-02-13:
Yes i have tried that from another post but without luck. On/off, power and maybe voltage works but current behaves strange. Maybe the integration time for current needs to be longer than 1s? Ive not used any other method than manually entered the pin config in the menu and it automatically seem to load the driver