openshwprojects/OpenBK7231T_App #1082: RGB color accuracy compared to stock Tuya and ESPHome firmware

ryan-ronnander, 2024-02-14:
Just want to say I love this project and I'm happy to help where I can, even if that's just opening a few bug reports to hopefully improve OpenBeken in comparison to ESPHome and Tasmota.

I'm using some SYLANIA Smart+ RGBW bulbs which are controlled by PWM. When comparing different firmware, OpenBeken seems to perform the worst compared to stock Tuya firmware and ESPHome. 

I created a video to hopefully try and show the differences between the firmware running on the same bulbs:

<div align="left">
      [image]
</div>

Somewhat longer video, but the reds being fairly off between the 3:00 - 4:00 minute mark are noticeable in the video. Of course, I'd throw a Tasmota comparison in as well, but the lack of Tasmota support is the reason I'm here :laughing:

openshwprojects, 2024-02-14:
Thank you for pointing that out. We have some kind of RGB calibration tool created by contributor: https://github.com/openshwprojects/OpenBK7231T_App/blob/main/docs/led_gamma_control.md
Do you think that it may help? Maybe we need to apply some kind of multiplier to colors by default to make red more red?

stromrickard, 2024-04-17:
Did you find any solution to improve this? I've got a Nedis RGBW LED-strip that i just flashed and i'm seeing the same thing after flashing it.

ryan-ronnander, 2024-04-17:
> Did you find any solution to improve this? I've got a Nedis RGBW LED-strip that i just flashed and i'm seeing the same thing after flashing it.

I've since flashed these bulbs to ESPHome.

ryan-ronnander, 2026-09-06:
I booted up an old bulb with OpenBeken on it still. Just retested with version 1.18.310. Colors look good, reds, etc.

closed 2026-09-06 by ryan-ronnander