openshwprojects/OpenBK7231T_App #1089: Stability problems on BL602 (LF686C20) using MQTT

CarlKuhligk, 2024-02-16:
**Describe the bug**
After sending a lot of MQTT messages, the device randomly disconnects from the WIFI and does not reconnect automatically. Switching it on again helps, but the instability persists.

**Firmware:**
- Version 1.17.463
- Device LF686C20 [details](https://www.elektroda.com/rtvforum/topic4024917.html)
- Chip/model: BT602
- Device config
```
"flags": "537134080",
"pins": {
    "3": "PWM;0",
    "4": "PWM;1",
    "20": "PWM;3",
    "21": "PWM;2"
  }
```

**To Reproduce**
Steps to reproduce the behavior:
1. switch on the device
2. send a lot of MQTT messages from node red
3. wait until the device has disappeared from WIFI.

**Logs**
```
Info:MAIN:Time 75, idle 0/s, free 108584, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 76, idle 0/s, free 108280, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 3/21 
Info:MAIN:Time 77, idle 0/s, free 108584, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 78, idle 0/s, free 108584, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 79, idle 0/s, free 108584, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MAIN:Time 80, idle 0/s, free 108584, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/Dimmer
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/HSBColor1
Info:MQTT:Publishing val 100 to livingroom_mini_bar_1/led_dimmer/get retain=0
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:MQTT:Publishing val FF1500 to livingroom_mini_bar_1/led_basecolor_rgb/get retain=0
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/HSBColor2
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/Power
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:CMD:tasCmnd POWER (Power) received with args 1
Info:MQTT:Publishing val 1 to livingroom_mini_bar_1/led_enableAll/get retain=0
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:MAIN:Time 81, idle 0/s, free 108584, MQTT 1(1), bWifi 1, secondsWithNoPing -1, socks 2/21 
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/Dimmer
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/HSBColor1
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/HSBColor2
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/Power
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:CMD:tasCmnd POWER (Power) received with args 1
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/Dimmer
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/HSBColor1
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/HSBColor2
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/Power
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Info:CMD:tasCmnd POWER (Power) received with args 1
Info:MQTT:Publishing val (128 bytes) to stat/livingroom_mini_bar_1/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic cmnd/livingroom_mini_bar_1/Power
Info:CMD:tasCmnd POWER (Power) received with args 0
Info:MQTT:Publishing val {"Dimmer":100,"Fade":"OFF","Speed":1,"LedTable":"ON","Color":"0,0,0","HSBColor":"5,100,100","Channel":[0,0,0],"POWER":"OFF"} to stat/livingroom_mini_bar_1/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic livingroom_mini_bar_1/led_dimmer/get
Info:MQTT:MQTT client in mqtt_incoming_publish_cb topic livingroom_mini_bar_1/led_basecolor_rgb/get
```
The device then does not respond for some time. Sometimes it re-establishes the connection to the mqtt server because an mqtt timeout triggers an action in the device. Without having opened the web interface of the device (after a restart), the device closes the WIFI connection and remains dead.

What can I do to further analyse the behaviour of the device? Does it make sense to log the device via the serial interface?

CarlKuhligk, 2024-02-16:
## Crashlog

I logged the output now via serial connection.
I will try again and see if it crashes at the same point every time.

<details>
<summary>Click to expand</summary>

```
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:CMD:tasCmnd POWER (power) received with args 1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:CMD:tasCmnd POWER (power) received with args 1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:CMD:tasCmnd POWER (power) received with args 1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:CMD:tasCmnd POWER (power) received with args 1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:CMD:tasCmnd POWER (power) received with args 1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:Publishing val (131 bytes) to stat/obl42BF98D4/RESULT retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:MQTT:MQTT will reconnect soon to fix ERR_MEM errors
Info:MQTT:MQTT has pending reconnect in 4
Info:MQTT:Publishing val 2 to obl42BF98D4/sockets retain=0
Info:MQTT:Publishing val FFD800 to obl42BF98D4/led_basecolor_rgb/get retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
Info:CMD:UART received:
Info:MAIN:Time 33, idle 0/s, free 106160, MQTT 1(2), bWifi 1, secondsWithNoPing -1, socks 2/21
Info:MQTT:MQTT has pending reconnect in 3
Info:MQTT:Publishing val -39 to obl42BF98D4/rssi retain=0
Error:MQTT:Publish err: ERR_MEM aka -1
[... 1823 lines cut ...]
Info:CMD:UART received:
Info:MAIN:Time 373, idle 0/s, free 89288, MQTT 1(18), bWifi 1, secondsWithNoPing 304, socks 5/21
Info:MQTT:MQTT has pending reconnect in 3
Info:CMD:UART received:
Info:MAIN:Time 374, idle 0/s, free 89288, MQTT 1(18), bWifi 1, secondsWithNoPing 305, socks 5/21
Info:MQTT:MQTT has pending reconnect in 2
Info:CMD:UART received:
Info:MAIN:Time 375, idle 0/s, free 89288, MQTT 1(18), bWifi 1, secondsWithNoPing 306, socks 5/21
Info:MQTT:MQTT has pending reconnect in 1
Info:CMD:UART received:
Info:MAIN:Time 376, idle 0/s, free 89288, MQTT 1(18), bWifi 1, secondsWithNoPing 307, socks 5/21
Info:MQTT:MQTT has pending reconnect in 0
Info:MQTT:MQTT will now do a forced reconnect
Info:CMD:UART received:
Info:MAIN:Time 377, idle 0/s, free 89288, MQTT 1(18), bWifi 1, secondsWithNoPing 308, socks 5/21
Info:CMD:UART received:
Info:MAIN:Time 378, idle 0/s, free 89288, MQTT 0(18), bWifi 1, secondsWithNoPing 309, socks 5/21
Info:CMD:UART received:
Info:MAIN:Time 379, idle 0/s, free 89288, MQTT 0(18), bWifi 1, secondsWithNoPing 310, socks 5/21
Info:MQTT:mqtt_userName ###mqtt user###
mqtt_pass ###mqtt password###
mqtt_clientID obl42BF98D4
mqtt_host ###mqtt host###
```
</details>

openshwprojects, 2024-02-16:
We've seen something similiar on BK7231. It was fixed by @valeklubomir  with LWIP update. Maybe we need to port changes to BL602...
https://github.com/openshwprojects/OpenBK7231T/pull/11/files

CarlKuhligk, 2024-02-18:
I would higly appreciate a similar fix. It would make this fantastic project much more usable.

vizualjack, 2024-02-19:
I would also welcome this fix very much. I hope you can implement it soon :D

openshwprojects, 2024-04-05:
We have recently added watchdog for BL602 so it should help mitigate potential problems before a final solution is ready.